ros2/ros2cli #856: Log file is not named after node name when running an `rclpy` based process

sea-bass, 2023-09-26:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Latest binaries on Humble
- DDS implementation:
  - Cyclone DDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
When I run the following command:

```
ros2 run demo_nodes_py talker --ros-args -r __node:=my_talker
```

#### Expected behavior

I would expect looking into my ROS logs folder, e.g., `~/.ros/log` and seeing a log file named after the

e.g. `my_talker_96363_1695751593253.log`

#### Actual behavior

However, I am instead seeing a generic Python name: `python3_96363_1695751593253.log`

This makes it very inconvenient to search for logs by node name, which should be the case.

#### Additional information

Running `ros2 node list` shows this node name correctly; it's just the logger that is wrong:

```
$ ros2 node list
/my_talker
```

Additionally, this does not happen with `rclcpp` based nodes.

For example

```
ros2 run demo_nodes_cpp talker --ros-args -r __node:=my_talker
```

yields the following log file name:

```
talker_96405_1695751970731.log
````

----

## Feature request

The node name should actually be used in the log file names created from `rclpy` nodes, as described in the "Expected Behavior" section above.

fujitatomoya, 2023-09-27:
ROS 2 can have multiple nodes in the same process space, so node name would not be suitable for log file name.

current implementation uses program name for the log file name,

https://github.com/ros2/rcl_logging/blob/014239dd80ab3de2a847af80f23d24da6438ecb7/rcl_logging_spdlog/src/rcl_logging_spdlog.cpp#L156

and,

https://github.com/ros2/rcutils/blob/f078c8c94885c94809d5d185822cf8ae8d4b795f/include/rcutils/process.h#L55

i think using program name is fine (`rclcpp` case) but once it comes to `python`, it returns `python3`.

something we can do is replace program name with `argv[1]: script name`? any thoughts?

sea-bass, 2023-09-27:
I see -- thank you! So each file is specific to a process? In that case, I think using the script name could work as a first pass.

Ideally, there would also be a way to configure the logger name in `ros2 run` / `ros2 launch` as well, though?

MichaelOrlov, 2023-10-22:
@fujitatomoya As regards
> something we can do is replace program name with argv[1]: script name? any thoughts?

Could you please clarify what you mean by that or in what place we could potentially do this?

MichaelOrlov, 2023-10-22:
I would consider adding one extra parameter **const char * program_name** to the 
```
rcl_logging_ret_t
rcl_logging_external_initialize(const char * program_name, const char * config_file, rcutils_allocator_t allocator);
```
and to the 
```
rcl_ret_t
rcl_logging_configure_with_output_handler(
  const char * program_name,
  const rcl_arguments_t * global_args,
  const rcl_allocator_t * allocator,
  rcl_logging_output_handler_t output_handler)
```
To be able to provide `program_name` explicitly from the upper layer.
It could be `null_ptr` by default and we can fallback to the `rcutils_get_executable_name(..)` call if it is null_ptr as we currently do.

fujitatomoya, 2023-10-24:
> Could you please clarify what you mean by that or in what place we could potentially do this?

```bash
root         388     385  3 20:20 pts/2    00:00:01 /usr/bin/python3 /root/ros2_ws/colcon_ws/install/demo_nodes_py/lib/demo_nodes_py/talker
```

- before is `python3_388_1698117640083.log`
- after is `talker_388_1698117640083.log`

MichaelOrlov, 2023-10-24:
@fujitatomoya Sorry still didn't get how you replaced the program name with argv[1] ?
How to do this with `ros2cli` API and extensions?

fujitatomoya, 2023-10-24:
@MichaelOrlov oh, that is just an idea to replace `python3` with `python script name`(argv[1]) via `rcutils_get_executable_name` if that is `python3`, so we cannot do that now. i do not think this is an good idea either, so i was thinking probably we can use global argument as well.

iuhilnehc-ynos, 2023-10-24:
another alternative solution, thoughts?

- rcutils

```c
src/logging.c

	static char g_rcutils_logging_custom_executable_name[RCUTILS_FILENAME_LEN];
	
	rcutils_ret_t rcutils_logging_set_executable_name(const char *);
	
	char * rcutils_logging_get_executable_name();
		get g_rcutils_logging_custom_executable_name
		check g_rcutils_logging_custom_executable_name if it's necessary to call `rcutils_get_executable_name`
```

- rcl_logging_spdlog
	use `rcutils_logging_get_executable_name` instead of `rcutils_get_executable_name` in `rcl_logging_spdlog/src/rcl_logging_spdlog.cpp`

- rclpy

```cpp
	rclpy/rclpy/src/rclpy/logging.cpp

	logging_configure
		to call rcutils_logging_set_executable_name(basename) before `rcl_logging_configure_with_output_handler`
		
		// to set the executable name
		try
		{
			py::module main_module = py::module::import("__main__");
			std::string file_name = main_module.attr("__file__").cast<std::string>();
			printf("outArray:%s\n", fs::path(file_name).filename().c_str());
			// to call `rcutils_logging_set_executable_name`
		}
		catch (const std::exception& ex)
		{
			RCUTILS_SAFE_FWRITE_TO_STDERR("rclpy failed to get the script file name");
			RCUTILS_SAFE_FWRITE_TO_STDERR(ex.what());
			RCUTILS_SAFE_FWRITE_TO_STDERR("\n");
		}
```

fujitatomoya, 2023-12-13:
@sea-bass could you try,

- https://github.com/ros2/rcl_logging/pull/109
- https://github.com/ros2/rcl/pull/1127

it works in my local environment.

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_py talker --ros-args --log-file-name foobar
[INFO] [1702509980.781646208] [talker]: Publishing: "Hello World: 0"
...
root@tomoyafujita:~/.ros/log# ls -lt
total 328
-rw-r--r-- 1 root root    0 Dec 13 15:26  foobar_585207_1702509979643.log
```

sea-bass, 2023-12-19:
> @sea-bass could you try,
> 
> * [add file_name_prefix parameter to external log configuration. rcl_logging#109](https://github.com/ros2/rcl_logging/pull/109)
> * [support `--log-file-name` to ros args. rcl#1127](https://github.com/ros2/rcl/pull/1127)
> 
> it works in my local environment.
> 
> ```shell
> root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_py talker --ros-args --log-file-name foobar
> [INFO] [1702509980.781646208] [talker]: Publishing: "Hello World: 0"
> ...
> root@tomoyafujita:~/.ros/log# ls -lt
> total 328
> -rw-r--r-- 1 root root    0 Dec 13 15:26  foobar_585207_1702509979643.log
> ```

This is great -- thank you for the update!

Would the path forward to specify this from launch files to similarly add to the ROS args using the existing launch APIs?

fujitatomoya, 2023-12-19:
@sea-bass 

> Would the path forward to specify this from launch files to similarly add to the ROS args using the existing launch APIs?

yeah, it should be, that was the intention. along with other arguments such as `--log-level`.

i think we need to add those test cases for `launch` and `ros2cli`, see https://github.com/ros2/rcl/pull/1127#discussion_r1425992174

fujitatomoya, 2024-01-04:
@clalancette i got a few follow-up PRs to close this issue. could you review these?

- https://github.com/ros2/launch_ros/pull/387
- https://github.com/ros2/rcl/pull/1130
- https://github.com/ros2/ros2_documentation/pull/4087

closed 2024-01-08 by clalancette