ros2/ros2cli #857: `ros2 action send_goal` shows incorrect goal UUID

alsora, 2023-09-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 
- Installation type:
  - sources
- Version or commit hash:
  - Galactic
- DDS implementation:
  - Fast-DDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Send a ROS 2 action goal from command line to an action server running in a C++ application.
The ROS 2 CLI prints a goal id when the goal is accepted.
Have the action server print the UUID of the goal using
```
rclcpp_action::to_string(goal_handle->get_goal_id())
```

#### Expected behavior

The UUID printed by the C++ action server and the UUID printed by the Python action client should match.

#### Actual behavior

The UUIDs are very similar but they do not match.
For example, the CLI shows:
```
Goal accepted with ID: b10a7c4d52564eafa06962e25eefe297
```

while the C++ app shows
```
b1a7c4d52564eafa06962e25eefe297
```

Note how the two string representations are very similar, but different.

The same applies to many other goals. It looks like there are always a couple of extra zeros added in the Python representation.

clalancette, 2023-09-28:
Can you please check with Rolling or Humble?  I can't seem to make it happen with Rolling, but I may just be getting "lucky" and not generating UUIDs with 0s at the right spots.

Ryanf55, 2024-02-25:
Trying to print the UUID with [uuid_unparse](https://linux.die.net/man/3/uuid_unparse) results in a different value than what is reported in the CLI on `humble`. Would you like me to provide a minimum reproducible example?

fujitatomoya, 2024-02-26:
@Ryanf55 yes please. for the information, does this also happen with `rolling`?

Ryanf55, 2024-02-26:
The code it's in is tied to humble. I'll extract a minimum example out and report test results on `humble` and `iron`.

clalancette, 2026-11-14:
No follow-up here, so I'm going to close this out for now.  Please feel free to reopen if this is still a problem for you.

closed 2026-11-14 by clalancette