ros2/ros2cli #869: ros2 topic {bw|delay|hz} don't support --no-daemon

scpeters, 2023-11-03:
## Feature request

#### Feature description

Support for the `--no-daemon` command-line argument was added to several of the `ros2 topic` verbs in https://github.com/ros2/ros2cli/pull/514, but it is not yet provided for the `bw`, `delay` or `hz` commands. It's also not available for `pub`, but I'm not sure if that matters.

#### Implementation considerations

fujitatomoya, 2023-11-04:
I think that question is we need to `NodeStrategy` for these command options.

> not yet provided for the bw, delay or hz commands. It's also not available for pub

I believe that all these sub-commands are used when we already know the topic is available via `ros2 topic list` or `find`, and then those sub-commands actually need to connect to the endpoint (not discovery configuration or graph). so i do not really think of the use case even for these...

any thoughts?

scpeters, 2023-11-04:
ok, please close this issue if it is not valid. I don't really understand how the tools work on the inside; I just noticed an asymmetry in the CLI flags for some of the verbs and thought they could be made more consistent

fujitatomoya, 2023-11-05:
@scpeters thanks for sharing your thoughts. i will close this for now, but there would be other opinion, and feel free to reopen!

closed 2023-11-05 by fujitatomoya

clalancette, 2023-11-05:
Yeah, it turns out that we basically use the daemon for caching discovery information, so that commands like `ros2 topic list` return all of the topics in the system.  In the case of commands where we don't need everything in the system, we don't use the daemon, and in that case we also don't need the `--no-daemon` command-line argument.  I believe that applies to all of `bw`, `delay`, and `hz`; they will all wait around for a particular topic, and hence don't need to use the daemon at all.

All of that is to say that I don't think we need to do anything here.  But thanks for bringing it up.