ros2/ros2cli #885: ROS 2 daemon process ignores SIGTERM

cottsay, 2024-03-05:
## Bug report

**Required Info:**

- Operating System:
  - Fedora 39
- Installation type:
  -  source
- Version or commit hash:
  - d3cfbd7
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
$ ros2 daemon start
$ ps aux | grep ros2
cottsay   736254  0.2  0.4 1859404 78720 pts/26  Sl   13:31   0:00 /usr/bin/python3 -c from ros2cli.daemon.daemonize import main; main() --name ros2-daemon --ros-domain-id 0 --rmw-implementation rmw_fastrtps_cpp
$ kill -TERM 736254
$ ros2 daemon status
The daemon is running
```

#### Expected behavior
Daemon process gracefully stops upon receiving `SIGTERM`.

#### Actual behavior
Daemon process ignores the signal.

#### Additional information

fujitatomoya, 2024-03-06:
i cannot figure out why `SIGINT` and `SIGKILL` just works fine but `SIGTERM`... any thoughts?

```bash
### does not work
kill -TERM 12280
### works
kill -KILL 12280
kill -INT 12280
```

clalancette, 2024-03-06:
I spent a bit of time digging into this.  I have some idea of what is going on here.

I think we have two bugs.  The first bug is that https://github.com/ros2/ros2cli/blob/d3cfbd7612b5d2707b16da1b900dc772dd88a61c/ros2cli/ros2cli/daemon/__init__.py#L128-L129 needs to also check `rclpy.ok`, since `rclpy` is handling the signals by default.

The second bug is a bit trickier.  It turns out that when we are sitting idle, we will never, ever leave `server.handle_request()` in that loop.  That `handle_request()` uses a `select` statement underneath, but the timeout for that select is set to `2*60*60`, i.e. two hours.  To fix this, I think we need to have two timeouts; one that is the timeout for checking whether we should quit due to a signal (set to 100 milliseconds or something like that), and one which is the overall timeout which defines when the daemon should quit due to inactivity.  I'll give a shot at implementing that.

clalancette, 2024-03-06:
All right, see #887 for the fixes.

PR #887 (2024-03-08): Fix the SIGTERM handling in the ros2 daemon.
There are 2 fixes needed here:
1.  Make sure to check rclpy.ok() in the overall loop for handling requests.  That's because rclpy is handling the signals by default.
2.  Realize that server.handle_request() uses the set timeout as a timeout to select(), essentially.  Because of that, we instead set the timeout to a short value (200 milliseconds), and do the overall timeout checking ourselves.

Note that because we are waking up more often, this will cause the ros2 daemon to use more CPU than before.  But it should be negligible overall.

This should [... cut]
closed 2024-03-08 by clalancette