ros2/ros2cli #889: echo `--spini-time` option doesn't exit after the given duration.

Jay-Waves, 2024-03-17:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04
  - 
- Installation type: binaries
  - 
- Version or commit hash: ros2-humble
  - 
- DDS implementation: Fast-DDS? maybe
  - 
- Client library (if applicable): N/A
  - 

#### Steps to reproduce issue

```
ros2 topic echo --spin-time 10 --no-daemon /cmd_vel_or_whatever_not_published
``` 

#### Expected behavior

After spinning and waiting for 10 secs, `ros2 topic echo` should return:
```
WARNING: topic [/cmd_vel] does not appear to be published yet
Could not determine the type for the passed topic 
```

#### Actual behavior

But `ros2 topic echo` just hung and didnot end itself for long time, which blocked my script.

#### Additional information

I tried `ros2 topic echo --spin-time 0.5 /cmd_vel`, and it ended itself and exit. But any other spin-time cloud make `echo` long time blocked.

I'm from #253, so i add `--no-daemon`, but not worked.

----

## Feature request

#### Feature description

#### Implementation considerations

fujitatomoya, 2024-03-17:
@Jay-Waves thanks for creating the issue.

so far, i cannot reproduce this problem with any current distros.

```bash
tomoyafujita@~/DVT >docker images | grep ros
ros                                                                                    rolling                   c6da1abec4d8   11 days ago     778MB
ros                                                                                    iron                      22184b5715a7   11 days ago     777MB
ros                                                                                    humble                    47cf82a0a3b2   11 days ago     752MB

tomoyafujita@~/DVT >docker run -it --rm ros:rolling
root@890ca654691c:/# source /opt/ros/rolling/setup.bash
root@890ca654691c:/# time ros2 topic echo --spin-time 10 --no-daemon /cmd_vel_or_whatever_not_published
WARNING: topic [/cmd_vel_or_whatever_not_published] does not appear to be published yet
Could not determine the type for the passed topic

real    0m10.393s
user    0m0.249s
sys     0m0.052s

tomoyafujita@~/DVT >docker run -it --rm ros:iron
root@11953e90cbbd:/# source /opt/ros/iron/setup.bash
root@11953e90cbbd:/# time ros2 topic echo --spin-time 10 --no-daemon /cmd_vel_or_whatever_not_published
WARNING: topic [/cmd_vel_or_whatever_not_published] does not appear to be published yet
Could not determine the type for the passed topic

real    0m10.389s
user    0m0.246s
sys     0m0.041s

tomoyafujita@~/DVT >docker run -it --rm ros:humble
root@99ebf28bbb7a:/# source /opt/ros/humble/setup.bash
root@99ebf28bbb7a:/# time ros2 topic echo --spin-time 10 --no-daemon /cmd_vel_or_whatever_not_published
WARNING: topic [/cmd_vel_or_whatever_not_published] does not appear to be published yet
Could not determine the type for the passed topic

real    0m10.496s
user    0m0.228s
sys     0m0.052s
```

please also see, https://github.com/ros2/ros2cli/issues/248#issuecomment-2002040281

Jay-Waves, 2024-03-22:
I apologize, as it appears the issue lies with the management of my child processes.

Thank you for your patience. ros2cli a nice tool yet. :)

closed 2024-03-22 by Jay-Waves