ros2/ros2cli #892: Several lines from _local_setup_util_sh.py contain this: E501 line too long

ionutnechita-intel, 2024-04-02:
OS:
ROS2 rolling - 02 April 2024
Ubuntu 24.04 Alpha LTS

This line have issue with flake8 and pycodestyle:

```
cat  _local_setup_util_sh.py
...
FORMAT_STR_INVOKE_SCRIPT = 'COLCON_CURRENT_PREFIX="{prefix}" _colcon_prefix_sh_source_script "{script_path}"'  # noqa: E501
FORMAT_STR_REMOVE_LEADING_SEPARATOR = 'if [ "$(echo -n ${name} | head -c 1)" = ":" ]; then export {name}=${{{name}#?}} ; fi'  # noqa: E501
FORMAT_STR_REMOVE_TRAILING_SEPARATOR = 'if [ "$(echo -n ${name} | tail -c 1)" = ":" ]; then export {name}=${{{name}%?}} ; fi'  # noqa: E501

```

I append  noqa: E501 to bypass this errors.

fujitatomoya, 2024-04-02:
I am not sure if here is the appropriate repository to track this issue, imo `_local_setup_util_sh.py` is generated by `colcon_core`?

CC: @clalancette @cottsay

clalancette, 2024-04-02:
Well, it is not the appropriate repository, but it is generated by ament instead: https://github.com/ament/ament_package/blob/35b15ae4d8c5e1dbb6ab622ff95d046aa3ca1b28/ament_package/template/prefix_level/_local_setup_util.py#L45-L54

That said, we don't regularly run flake8 and pycodestyle over the *results* of this output.

What I'm going to do here is to close this one out (since we can't move issues between organizations).  @ionutnechita-intel Feel free to reopen this against https://github.com/ament/ament_package , but we will likely mark it as "help wanted" over there.

closed 2024-04-02 by clalancette

ionutnechita-intel, 2024-04-16:
I reopen this against in: https://github.com/ament/ament_package

Issue: https://github.com/ament/ament_package/issues/146