ros2/ros2cli #896: ros2action.api ActionTypeCompleter results in traceback

Algias, 2024-04-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binaries
- Version or commit hash:
  - humble
- DDS implementation:
  - N/A
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```
from ros2action.api import ActionTypeCompleter
completer = ActionTypeCompleter()
completer("", None)
```

Similarly
```
from ros2action.api import action_type_completer
action_type_completer()
```

#### Expected behavior
`ActionTypeCompleter` should be a callable that returns a list of actions available from packages if an action name is not passed in. 

#### Actual behavior
A traceback occurs indicating the bug
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2action/api/__init__.py", line 92, in __call__
    return action_type_completer()
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2action/api/__init__.py", line 78, in action_type_completer
    for package_name, action_names in get_action_interfaces.items():
AttributeError: 'function' object has no attribute 'items'
```
Or in the case it is used in a verb with tab-complete, no items are displayed.
#### Additional information
I believe the bug is on this line which is present in rolling as of the indicated commit:
https://github.com/ros2/ros2cli/blob/ce945cd70d83a9ff41f08594e40ea96a0e6ac308/ros2action/ros2action/api/__init__.py#L78

`ServiceTypeCompleter` does not have this issue
https://github.com/ros2/ros2cli/blob/ce945cd70d83a9ff41f08594e40ea96a0e6ac308/ros2service/ros2service/api/__init__.py#L95

clalancette, 2024-04-10:
> I believe the bug is on this line which is present in rolling as of the indicated commit:

Yep, that looks like it is the problem.  Thanks for tracking it down.  Would you be interested in opening a pull request to fix it?  Then we can review it and get it in.

fujitatomoya, 2024-04-18:
i happened to meet this problem, fix is ready https://github.com/ros2/ros2cli/pull/898

fujitatomoya, 2024-04-24:
closing in favor of https://github.com/ros2/ros2cli/pull/898, all backports (jazzy, iron and humble) completed as well.

closed 2024-04-24 by fujitatomoya