ros2/ros2cli #906: ros2doctor has incorrect bitmask for MULTICAST on macOS 14.4

srmainwaring, 2024-06-03:
## Bug report

**Required Info:**

- Operating System:
  - macOS Sonoma 14.4.1
- Installation type:
  - source
- Version or commit hash:
  - `git log commit 58b61c98378fa49a4a164450f1d5222bde2e4f50 (HEAD -> humble, tag: 0.18.10, origin/humble)`
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - `ros2doctor`

#### Steps to reproduce issue

```
ros2 doctor
```

#### Expected behavior

No output, as multicast is supported on loopback:

```bash
 % ifconfig
lo0: flags=8049<UP,LOOPBACK,RUNNING,MULTICAST> mtu 16384
	options=1203<RXCSUM,TXCSUM,TXSTATUS,SW_TIMESTAMP>
	inet 127.0.0.1 netmask 0xff000000
	inet6 ::1 prefixlen 128 
	inet6 fe80::1%lo0 prefixlen 64 scopeid 0x1 
	nd6 options=201<PERFORMNUD,DAD>
...
```

#### Actual behavior

```bash
/Users/rhys/Code/ros2/humble/ros2-humble/build/ros2doctor/ros2doctor/api/network.py: 166: UserWarning: No multicast IP address is found.
```

#### Additional information

Bitmask used in `ros2doctor` defined here:

https://github.com/ros2/ros2cli/blob/58b61c98378fa49a4a164450f1d5222bde2e4f50/ros2doctor/ros2doctor/api/network.py#L30-L50

Bitmask `#defines` for net interfaces in macOS 14.4:

```c
/* /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/System/Library/Frameworks/Kernel.framework/Versions/A/Headers/net/if.h */

#define IFF_UP          0x1             /* interface is up */
#define IFF_BROADCAST   0x2             /* broadcast address valid */
#define IFF_DEBUG       0x4             /* turn on debugging */
#define IFF_LOOPBACK    0x8             /* is a loopback net */
#define IFF_POINTOPOINT 0x10            /* interface is point-to-point link */
#define IFF_NOTRAILERS  0x20            /* obsolete: avoid use of trailers */
#define IFF_RUNNING     0x40            /* resources allocated */
#define IFF_NOARP       0x80            /* no address resolution protocol */
#define IFF_PROMISC     0x100           /* receive all packets */
#define IFF_ALLMULTI    0x200           /* receive all multicast packets */
#define IFF_OACTIVE     0x400           /* transmission in progress */
#define IFF_SIMPLEX     0x800           /* can't hear own transmissions */
#define IFF_LINK0       0x1000          /* per link layer defined bit */
#define IFF_LINK1       0x2000          /* per link layer defined bit */
#define IFF_LINK2       0x4000          /* per link layer defined bit */
#define IFF_ALTPHYS     IFF_LINK2       /* use alternate physical connection */
#define IFF_MULTICAST   0x8000          /* supports multicast */
```

clalancette, 2024-06-03:
I think probably the correct fix here is to switch to using psutil to get this information for us; it has all of it in a cross-platform way.

Can you give the code in https://github.com/ros2/ros2cli/tree/clalancette/revamp-doctor-network a whirl and see if it is better on macOS (that branch should work with Rolling).  If that is better, then I can test it out on Windows and open a PR for it.

srmainwaring, 2024-06-06:
> Can you give the code in https://github.com/ros2/ros2cli/tree/clalancette/revamp-doctor-network a whirl and see if it is better on macOS (that branch should work with Rolling). If that is better, then I can test it out on Windows and open a PR for it.

Thanks @clalancette - this works well on macOS 14.5.

clalancette, 2024-06-11:
> Thanks @clalancette - this works well on macOS 14.5.

Awesome, thanks for testing.  I've opened up #910 now.

PR #910 (2024-06-13): Revamp how we get network information in ros2doctor.
We already depend on psutil, which can give us information for ros2doctor in a cross-platform way.  Lean more heavily into using this, which should help us deal with cross-platform a little more smoothly.

I will note that this slightly changes the output of ros2doctor on Linux.  Prior to this PR, the flags output would look something like:

```
flags        : 4163<MULTICAST,BROADCAST,UP,RUNNING>
```

After this PR, the output looks like this:

```
flags        : UP,BROADCAST,RUNNING,MULTICAST
```

That is, we lose the decimal bitmask associ [... cut]
closed 2024-06-13 by clalancette