ros2/ros2cli #909: removeprefix breaks compatibility with source builds on Nvidia Jetson

robmasocco, 2024-06-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
  - Nvidia JetPack 5.1.1
  - Jetson Linux 35.3.1
- Installation type:
  - from source
- Version or commit hash:
  - 0c8ae6d976e2fdca1b109f5cb5e330691fe26fa0
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
```
ros2 topic list
```

#### Expected behavior
Issued command works.

#### Actual behavior
```
Traceback (most recent call last):
  File "/opt/ros/humble/install/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.10', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/install/lib/python3.8/site-packages/ros2cli/cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/humble/install/lib/python3.8/site-packages/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "/opt/ros/humble/install/lib/python3.8/site-packages/ros2topic/verb/list.py", line 55, in main
    with NodeStrategy(args) as node:
  File "/opt/ros/humble/install/lib/python3.8/site-packages/ros2cli/node/strategy.py", line 35, in __init__
    spawn_daemon(args)
  File "/opt/ros/humble/install/lib/python3.8/site-packages/ros2cli/node/daemon.py", line 148, in spawn_daemon
    fdlimit = int(line.removeprefix(string_to_find).strip())
AttributeError: 'str' object has no attribute 'removeprefix'
```

#### Additional information
Humble Hawksbill source builds are possible on Nvidia Jetson boards using JetPack 5.x, which is based on Ubuntu 20.04.
JetPack 5.x's Python version is 3.8, but `removeprefix` was added in Python version 3.9.

tonynajjar, 2024-11-15:
> Humble Hawksbill source builds are possible on Nvidia Jetson boards using JetPack 5.x, which is based on Ubuntu 20.04.
JetPack 5.x's Python version is 3.8, but removeprefix was added in Python version 3.9.

I also managed to build Jazzy on Jetpack 5.x and have the same issue. (I'm blocked from upgrading to Jetpack 6)
@clalancette would you oppose merging a compatibility PR for Jazzy?

clalancette, 2024-11-15:
> @clalancette would you oppose merging a compatibility PR for Jazzy?

For Jazzy?  I don't think we need it; Jazzy only targets Ubuntu 24.04 (Python 3.12) and Ubuntu 22.04 (Python 3.10).  Are you trying to build Jazzy on Ubuntu 20.04?

(Humble is a different story, since it targets Ubuntu 22.04 (Python 3.10) and Ubuntu 20.04 (Python 3.8))

tonynajjar, 2024-11-15:
> Are you trying to build Jazzy on Ubuntu 20.04?

(Unfortunately) yes. I'm stuck on Jetpack 5.x which has Ubuntu 20. I managed to build Jazzy on Ubuntu 20 with very few workarounds.
I understand that Jazzy does not officially support Ubuntu 20 but I think many Jetson users will be in my position and will want this fix. It's not super important for me though, I can live with a fix in my fork. Let's see if others comment on this thread

clalancette, 2024-11-15:
> (Unfortunately) yes. I'm stuck on Jetpack 5.x which has Ubuntu 20. I managed to build Jazzy on Ubuntu 20 with very few workarounds.

Oh wow, I'm pretty surprised about that.  I would have figured there would be tons of workarounds needed.

In that case, I'm OK with adding a fallback for `removeprefix`.  Note that I'd want to do this on `rolling` first, and then we could consider backporting to `jazzy` and `humble`.

closed 2024-12-17 by fujitatomoya