ros2/ros2cli #911: Trivial patch to fix misprint of action server

sbarber-dsi, 2024-06-13:
I found this bug in humble working with actions and pulled the latest and it's not fixed in rolling.
See ros2action/api/__init__.py:11

Code currently is:
    `node_fqn = '/'.join(node_ns) + node_name`

But it should be something like:
    `node_fqn = f'{node_ns}/{node_name}'`

e.g. for our GPS reset action
`ros2 action info <action_path>` currently prints:
`//v/2/2///a/u/t/o/n/o/m/y///s/e/n/s/o/r///g/p/snovatel`
but it should be
`/v22/autonomy/sensor/gps/novatel`

```
diff --git a/ros2action/ros2action/api/__init__.py b/ros2action/ros2action/api/__init__.py
index d0bf18d..f94866a 100644
--- a/ros2action/ros2action/api/__init__.py
+++ b/ros2action/ros2action/api/__init__.py
@@ -34,7 +34,7 @@ def get_action_clients_and_servers(*, node, action_name):
     node_names_and_ns = node.get_node_names_and_namespaces()
     for node_name, node_ns in node_names_and_ns:
         # Construct fully qualified name
-        node_fqn = '/'.join(node_ns) + node_name
+        node_fqn = f'{node_ns}/{node_name}'
 
         # Get any action clients associated with the node
         client_names_and_types = node.get_action_client_names_and_types_by_node(
```

christophebedard, 2024-06-13:
Thanks for reporting this! Would you be willing to open a pull request with this fix?

fujitatomoya, 2024-06-13:
rolling reproduces this issue, i think we also can add test case with namespace as below.

```shell
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run action_tutorials_cpp fibonacci_action_server --ros-args -r __ns:=/demo

root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 action list
/demo/fibonacci
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 action info /demo/fibonacci
Action: /demo/fibonacci
Action clients: 0
Action servers: 1
    //d/e/m/ofibonacci_action_server

```

closed 2024-09-05 by ahcorde