ros2/ros2cli #919: Allow ros2 topic echo output to be used in ros2 topic pub

Ryanf55, 2024-07-24:
## Feature request

#### Feature description

I want to use `ros2 topic echo`, copy the contents of a message in CLI, and run `ros2 topic pub` to publish the data without having to reformat it. 

Currently, you can echo a topic like this:
```
$ ros2 topic echo /geo_path --once
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: ''
poses:
- header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: ''
  pose:
    position:
      latitude: 1.0
      longitude: 2.0
      altitude: 3.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
- header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: ''
  pose:
    position:
      latitude: 4.0
      longitude: 5.0
      altitude: 6.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
---
```

Then, you need to tediously reformat it like this to publish:
```bash
$ ros2 topic pub /geo_path geographic_msgs/msg/GeoPath '{poses: [{pose: {position: {latitude: 1.0, longitude: 2.0, altitude: 3.0}}}, {pose: {position: {latitude: 4.0, longitude: 5.0, altitude: 6.0}}}]}'
```

#### Implementation considerations

Proposal: Allow publishing by yaml file like so:
`ros2 topic pub /geo_path geographic_msgs/msg/GeoPath --yaml my_message.yaml`

Where `my_message.yaml` is the direct YAML from ros2 topic echo.

I'm open to other alternatives such as `cat` operation on the yaml file and piping it to `xargs`. 
`cat my_message.yaml | xargs  -I{} ros2 topic pub /geo_path geographic_msgs/msg/GeoPath {}`

clalancette, 2024-07-24:
Does the `--flow-style` parameter to `ros2 topic echo` do what you want?

Ryanf55, 2024-07-24:
Yea, looks like it does. 
```
ros2 topic echo /geo_path --flow-style
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: ''
poses: [{header: {stamp: {sec: 0, nanosec: 0}, frame_id: ''}, pose: {position: {latitude: 1.0, longitude: 2.0, altitude: 3.0}, orientation: {x: 0.0, y: 0.0, z: 0.0, w: 1.0}}}, {header: {stamp: {sec: 0, nanosec: 0}, frame_id: ''}, pose: {position: {latitude: 4.0, longitude: 5.0, altitude: 6.0}, orientation: {x: 0.0, y: 0.0, z: 0.0, w: 1.0}}}]
---
```

Then, copy the content, type the command, add quotes for the message, and paste it in. 

```

ros2 topic pub /geo_path geographic_msgs/msg/GeoPath "header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: ''
poses: [{header: {stamp: {sec: 0, nanosec: 0}, frame_id: ''}, pose: {position: {latitude: 1.0, longitude: 2.0, altitude: 3.0}, orientation: {x: 0.0, y: 0.0, z: 0.0, w: 1.0}}}, {header: {stamp: {sec: 0, nanosec: 0}, frame_id: ''}, pose: {position: {latitude: 4.0, longitude: 5.0, altitude: 6.0}, orientation: {x: 0.0, y: 0.0, z: 0.0, w: 1.0}}}]"
```

If this is the intent of `flow-style`, perhaps the docs could make that clearer?

clalancette, 2024-07-24:
> If this is the intent of `flow-style`, perhaps the docs could make that clearer?

I'm not totally sure what the original intent of it was, but certainly an update to the documentation would be welcome.

fujitatomoya, 2024-07-25:
i think `--flow-stype` only helps the collection arrays in the block style,

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic pub /tf_static tf2_msgs/msg/TFMessage "transforms: [{header: {stamp: {sec: 1721935886, nanosec: 528868866}, frame_id: foo}, child_frame_id: bar, transform: {translation: {x: 1.0, y: 2.0, z: 3.0}, rotation: {x: -0.4747921762038255, y: -0.07596622270177095, z: 0.24006245183344296, w: 0.8433098728485138}}}]"

root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic echo --once /tf_static
transforms:
- header:
    stamp:
      sec: 1721935886
      nanosec: 528868866
    frame_id: foo
  child_frame_id: bar
  transform:
    translation:
      x: 1.0
      y: 2.0
      z: 3.0
    rotation:
      x: -0.4747921762038255
      y: -0.07596622270177095
      z: 0.24006245183344296
      w: 0.8433098728485138
---
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic echo --once --flow-style /tf_static
transforms: [{header: {stamp: {sec: 1721935886, nanosec: 528868866}, frame_id: foo}, child_frame_id: bar, transform: {translation: {x: 1.0, y: 2.0, z: 3.0}, rotation: {x: -0.4747921762038255, y: -0.07596622270177095, z: 0.24006245183344296, w: 0.8433098728485138}}}]
---
```

but it does not help if that is not the collection arrays,

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic pub /turtle1/cmd_vel geometry_msgs/msg/Twist "{linear: {x: 2.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.8}}"

root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic echo --once /turtle1/cmd_vel
linear:
  x: 2.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 1.8
---
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic echo --once --flow-style /turtle1/cmd_vel
linear:
  x: 2.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 1.8
---
```

> If this is the intent of flow-style, perhaps the docs could make that clearer?

so probably not for this use case...

> Then, copy the content, type the command, add quotes for the message, and paste it in.

this works even without `--flow-style` is specified,

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic pub /turtle1/cmd_vel geometry_msgs/msg/Twist "linear:
  x: 2.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 1.8
"
publisher: beginning loop
publishing #1: geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=2.0, y=0.0, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=1.8))

publishing #2: geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=2.0, y=0.0, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=1.8))
```

maybe we can add some suggestion about `--flow-style` for collection data,

> Proposal: Allow publishing by yaml file like so: ros2 topic pub /geo_path geographic_msgs/msg/GeoPath --yaml my_message.yaml

i think this is useful new feature during development and debugging, we can easily modify the certain data amount to play back by yaml file. we could use `rosbag2` but that requires us to modify the backend data base to modify the data contents.

mjcarroll, 2024-08-08:
Discussion from maintenance triage:  It was agreed that `--flow-style` is insufficient for what you are trying to accomplish.  This would be a very helpful feature for the reasons that you stated above.  There isn't any maintainer bandwidth to support creating these changes, but a PR would be happily accepted/reviewed.

fujitatomoya, 2024-08-30:
@Ryanf55 i got some spare time to develop this, if you can try https://github.com/ros2/ros2cli/pull/925, that would be appreciated!

Ryanf55, 2024-08-31:
Neat, I'm off on holiday weekend, but will review this when I get back Tuesday. Thanks for working on it!

closed 2025-02-07 by ahcorde