ros2/ros2cli #922: Ros2 Topic Echo Not Working

MichaelrMentele, 2024-08-02:
## Bug report

**Required Info:**

- Operating System: Ubuntu 24.04
- Installation type: deb binary 
- Version or commit hash: jazzy? Not sure how to get this, not seeing a version in the ROS2 cli help
- DDS implementation: Fast-RTPS
- Client library (if applicable): n/a

#### Steps to reproduce issue
https://docs.ros.org/en/jazzy/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Topics/Understanding-ROS2-Topics.html
```
ros2 run turtlesim turtle_teleop_key
ros2 run turtlesim turtlesim_node # seperate shell
ros2 topic echo /turtle1/cmd_vel # separate shell
```

#### Expected behavior
Should print out some kind of output for the turtles position and vector direction

#### Actual behavior
Nothing is printed out when I move the turtle

#### Additional information
Perhaps this is a symptom of this error: 
2024-08-02 10:50:18.217 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7001: open_and_lock_file failed -> Function open_port_internal

But I only see this when I run rqt.

MichaelrMentele, 2024-08-02:
Complete ROS n00b may be user error.

MichaelrMentele, 2024-08-02:
Yeah looks like a subscription issue -- the echo subscription is orphaned:

```
ros2 topic info /turtle1/cmd_vel
Type: geometry_msgs/msg/Twist
Publisher count: 1
Subscription count: 1
```
It's odd, i would have expected some kind of error message if it failed to connect. That's the secondary issue here. I wonder if this is all downstream of the transport error but why don't I see it when I run the command i.e. only with rqt?

MichaelrMentele, 2024-08-02:
So, I still had the issue after sourcing and opening a new shell but the issue went away once I killed the terminal application completely. 

Not sure if we want to leave this open, but after killing my terminal and all shells and reopening -- I'm good.

fujitatomoya, 2024-08-02:
@MichaelrMentele you can keep this open. actually i assigned myself to this, just in case i want to make sure this problem does not happen with my environment as well.

fujitatomoya, 2024-08-05:
i cannot reproduce this issue with `jazzy`: `ros-jazzy-rqt` 1.6.0-2noble.20240712.140504

[image]

closed 2024-08-05 by fujitatomoya