ros2/ros2cli #924: Empty ROS2 GetParameters service response if any of the parameter is declared but not defined

EwingKang, 2024-08-22:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 22.04
- Installation type:
  -  binaries
- Version or commit hash:
  -  IRON
- DDS implementation:
  -  N/A
- Client library (if applicable):
  -  ros cli (ros service)

Empty/ broken GetParameters service response if any of the parameter is declared but not defined.

#### Steps to reproduce issue
```c++ 
//my_node.cpp....

// declare parameters with default value
this->declare_parameter("lambda", 0.3, desc("lambda"));
this->declare_parameter("alpha", 5.0, desc("alpha"));

// This parameter is declared without default value causes problem
this->declare_parameter("help", rclcpp::PARAMETER_BOOL);

//...
```
Call the `DescribeParameters` service returns fine: 
```bash
ros2 service call /my_node/get_parameters rcl_interfaces/srv/GetParameters "{names: [alpha, beta]}"

# Response
requester: making request: rcl_interfaces.srv.GetParameters_Request(names=['alpha', 'beta'])

response:
rcl_interfaces.srv.GetParameters_Response(values=[rcl_interfaces.msg.ParameterValue(type=3, bool_value=False, integer_value=0, double_value=0.3, string_value='', byte_array_value=[], bool_array_value=[], integer_array_value=[], double_array_value=[], string_array_value=[]), rcl_interfaces.msg.ParameterValue(type=3, bool_value=False, integer_value=0, double_value=5.0, string_value='', byte_array_value=[], bool_array_value=[], integer_array_value=[], double_array_value=[], string_array_value=[])])
```

However, if any one of the requested parameter has no default parameter, the entire service call returns broken entry
```bash
ros2 service call /my_node/get_parameters rcl_interfaces/srv/GetParameters "{names: [alpha, beta, help]}"

# Response
waiting for service to become available...
requester: making request: rcl_interfaces.srv.GetParameters_Request(names=['alpha', 'beta', 'help'])

response:
rcl_interfaces.srv.GetParameters_Response(values=[])
```

#### Expected behavior
For parameters with default value, return normally; for parameters without default value, some sort of flag to mark the situation

#### Actual behavior
As described above

#### Additional information
I believe this is related to https://github.com/ros-visualization/rqt_reconfigure/issues/142

EwingKang, 2024-08-22:
This might be related to: https://github.com/ros2/rclcpp/issues/2512

fujitatomoya, 2024-08-22:
@EwingKang thanks for posting issue. a couple of things i want to confirm.

> Call the DescribeParameters service returns fine:

it looks like you are calling `GetParameters` service, is this typo?

> ros2 service call /my_node/get_parameters rcl_interfaces/srv/GetParameters "{names: [alpha, beta]}"

where is this `beta` defined? is this supposed to be `lamda`?

can you check the issue description? i just want to be sure the issue description.

EwingKang, 2024-08-22:
@fujitatomoya I'm terribly embarrassed by the mistake! I fixed the description.
Thanks for catching the error! It is exactly like what you imagine. Things were lost during copy and paste.
Please let me know if there's any confusion, thanks!

fujitatomoya, 2024-08-22:
@EwingKang thanks for checking this.

as you mentioned, this is similar with https://github.com/ros2/rclcpp/issues/2512 which explains specifically on `GetParameterTypes` but the point is inconsistent behavior against documentation.

Here it  complains about `GetParameter`, the behavior also does not match with https://docs.ros2.org/galactic/api/rcl_interfaces/srv/GetParameterTypes.html.

fujitatomoya, 2024-08-22:
Note: I think this issue can move to `rclcpp` repository since the service server is implemented in `rclcpp` that responds the service request via ros2cli.

sloretz, 2024-08-29:
Thank you for the report! Closing as a duplicate of https://github.com/ros2/rclcpp/issues/2512 🧇

closed 2024-08-29 by sloretz