ros2/ros2cli #932: :farmer: ros2cli pytest timeout in RHEL nightlies

Crola1702, 2024-09-16:
## Bug report

**Required Info:**

- Operating System:
  - RHEL 9
- Installation type:
  - Source
- Version or commit hash:
  - Rolling

#### Steps to reproduce issue
1. Run a build in [nightly-rhel-debug](https://ci.ros2.org/job/nightly_linux-rhel_debug/) or release
2. See test regression fail

#### Additional information

Reference build: https://ci.ros2.org/job/nightly_linux-rhel_release/2020/

Test failure:
* [ros2cli.pytest.missing_result](https://ci.ros2.org/job/nightly_linux-rhel_release/2020/testReport/(root)/ros2cli/pytest_missing_result/)

Log output:
```
timeout: 900.0s
timeout method: thread
timeout func_only: False
[1mcollecting ... [0m[1m
collecting 0 items                                                             [0m[1m
collected 31 items                                                             [0m

test/test_copyright.py [32m.[0m[32m                                                 [  3%][0m
test/test_daemon.py [32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m                                  [ 64%][0m
test/test_direct.py [32m.[0m[32m.[0m[32m.[0m[32m.[0m[32m                                                 [ 77%][0m
Fatal Python error: Segmentation fault

...

  File "/home/jenkins-agent/workspace/nightly_linux-rhel_release/ws/src/ros2/ros2cli/ros2cli/test/test_flake8.py", line 22, in test_flake8
    rc, errors = main_with_errors(argv=[])
  File "/home/jenkins-agent/workspace/nightly_linux-rhel_release/ws/install/ament_flake8/lib/python3.9/site-packages/ament_flake8/main.py", line 88, in main_with_errors
    report = generate_flake8_report(
  File "/home/jenkins-agent/workspace/nightly_linux-rhel_release/ws/install/ament_flake8/lib/python3.9/site-packages/ament_flake8/main.py", line 281, in generate_flake8_report
    report.report = style.check_files(paths)

...

Finished <<< ros2cli [15min 10s]	[ with test failures ]
```

Seems ros2cli flake8 is taking a lot of time now. Failing since: https://github.com/ros2/ros2cli/pull/929, but probably not the root cause

<details>
<summary>
Full log output:
</summary>

```
platform linux -- Python 3.9.18, pytest-6.2.2, py-1.10.0, pluggy-0.13.1
cachedir: /home/jenkins-agent/workspace/nightly_linux-rhel_release/ws/build/ros2cli/.pytest_cache
rootdir: /home/jenkins-agent/workspace/nightly_linux-rhel_release/ws/src/ros2/ros2cli, configfile: pytest.ini
plugins: launch-testing-3.6.1, ament-xmllint-0.18.1, ament-copyright-0.18.1, ament-pep257-0.18.1, ament-mypy-0.18.1, ament-flake8-0.18.1, ament-lint-0.18.1, timeout-1.4.2, mock-3.6.1, repeat-0.9.3, cov-4.0.0, colcon-core-0.17.0, rerunfailures-9.1.1
timeout: 900.0s
timeout method: thread
timeout func_only: False
[1mcollecting ... [0m[1m
collecting 0 items                                                             [0m[1m
collected 31 items                                                             [0m

test/test_copyright.py [32m.[0m[32m                                                 [  3%][0m
[... 270 lines cut ...]
  File "/home/jenkins-agent/workspace/nightly_linux-rhel_release/ws/src/ros2/ros2cli/ros2cli/test/test_flake8.py", line 22, in test_flake8
    rc, errors = main_with_errors(argv=[])
  File "/home/jenkins-agent/workspace/nightly_linux-rhel_release/ws/install/ament_flake8/lib/python3.9/site-packages/ament_flake8/main.py", line 88, in main_with_errors
    report = generate_flake8_report(
  File "/home/jenkins-agent/workspace/nightly_linux-rhel_release/ws/install/ament_flake8/lib/python3.9/site-packages/ament_flake8/main.py", line 281, in generate_flake8_report
    report.report = style.check_files(paths)
  File "/usr/lib/python3.9/site-packages/flake8/api/legacy.py", line 109, in check_files
    self._application.run_checks(paths)
  File "/usr/lib/python3.9/site-packages/flake8/main/application.py", line 271, in run_checks
    self.file_checker_manager.run()
  File "/usr/lib/python3.9/site-packages/flake8/checker.py", line 309, in run
    self.run_parallel()
  File "/usr/lib/python3.9/site-packages/flake8/checker.py", line 275, in run_parallel
    for ret in pool_map:
  File "/usr/lib64/python3.9/multiprocessing/pool.py", line 448, in <genexpr>
    return (item for chunk in result for item in chunk)
  File "/usr/lib64/python3.9/multiprocessing/pool.py", line 858, in next
    self._cond.wait(timeout)
  File "/usr/lib64/python3.9/threading.py", line 312, in wait
    waiter.acquire()

+++++++++++++++++++++++++++++++++++ Timeout ++++++++++++++++++++++++++++++++++++
```

</details>

clalancette, 2024-09-17:
I took a look into this, and this is almost certainly failing because of https://github.com/ros2/rclpy/pull/1349 .  Since that is such a large PR, it will be somewhat hard to track it down exactly, but with that reverted the failure stops happening for me.

fujitatomoya, 2024-09-18:
@clalancette @Crola1702 https://github.com/ros2/rclpy/pull/1359 revert PR.

fujitatomoya, 2024-09-19:
### Current Summary

There are 2 possible PRs to generate this RHEL issue.

- https://github.com/ros2/rclpy/pull/1349 (backport staging https://github.com/ros2/rclpy/pull/1359), that depends on https://github.com/ros2/rclpy/pull/1345
- https://github.com/ros2/rclpy/pull/1345 (backport merged https://github.com/ros2/rclpy/pull/1360)

@Crola1702 if possible, can you check with current rolling if this issue is still happening with RHEL? i just want to be sure if either or both of them are generating this issue.

CC: @clalancette @InvincibleRMC

InvincibleRMC, 2024-09-20:
Looking at [nightly](https://ci.ros2.org/job/nightly_linux-rhel_release/2023/testReport/junit/(root)/ros2cli/pytest_missing_result/) It still seems to be failing. So [Revert "Executors types (#1345)" rclpy#1360](https://github.com/ros2/rclpy/pull/1360)) was not enough.

fujitatomoya, 2024-09-20:
@InvincibleRMC thanks, then let's revert https://github.com/ros2/rclpy/pull/1359

fujitatomoya, 2024-09-20:
@InvincibleRMC both https://github.com/ros2/rclpy/pull/1359 and https://github.com/ros2/rclpy/pull/1360 are reverted, hopefully RHEL nightly comes back with green light 🤞

clalancette, 2024-09-21:
Since I was able to reproduce the issue, I'm pretty sure that https://github.com/ros2/rclpy/pull/1359 fixes this.  I'm going to close this issue out, but feel free to reopen if it reoccurs.

Crola1702, 2024-10-14:
This is happening consistently in [nightly_linux-rhel_release](https://ci.ros2.org/job/nightly_linux-rhel_release/2047/)

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
nightly_linux-rhel_release|2024-10-14|2024-10-04|15|8|53.33
nightly_linux-rhel_debug|2024-10-14|2024-10-04|14|7|50.0

Rhel release is consistent and Rhel debug is flaky

fujitatomoya, 2024-10-14:
@InvincibleRMC yes, https://github.com/ros2/rclpy/pull/1338 is the only commit since then. lets revert it. (I think that we should run all CI including RHEL debug for any type checking PR.)

InvincibleRMC, 2024-10-14:
> @InvincibleRMC yes, https://github.com/ros2/rclpy/pull/1338 is the only commit since then. lets revert it. (I think that we should run all CI including RHEL debug for any type checking PR.)

Sounds good. My apologies again.

fujitatomoya, 2024-10-17:
@InvincibleRMC @clalancette @Crola1702 

according to the experimental test with RHEL Debug/Release, https://github.com/ros2/rclpy/pull/1373#issuecomment-2418344932 says https://github.com/ros2/rclpy/pull/1363/files is the one that generated this regression as a result.

I just cannot explain why, if you have any idea, please let me know.

here is what i am gonna do, 1st priority is get RHEL back to be stable and green. although i cannot explain the logics why RHEL does not like https://github.com/ros2/rclpy/pull/1363/files. that is true and what happens. So i will go ahead to revert https://github.com/ros2/rclpy/pull/1363 and close https://github.com/ros2/rclpy/pull/1373. any objections?

InvincibleRMC, 2024-10-17:
That's very odd. Is there any chance you could run each of `roscli` tests by themselves? It seems to me `test_flake8.py` is the likely cause since that seems to be what the stack trace says but, it could be possible that another test is crashing and taking down flake8 with it mid run.

fujitatomoya, 2024-10-17:
@InvincibleRMC 
CC: @sloretz @clalancette 

We talked about this today. Although we are not fully sure what the root cause this is, it does seem there would be a bug somewhere else like flake8 or RHEL platform specific. with that and problem could be related to flake8 multi-threading, we decided to try flake8 in single-threaded configuration for CI https://github.com/ament/ament_lint/pull/505.

if https://github.com/ament/ament_lint/pull/505 comes back all green, probably we do not need to revert things now. i will keep my eyes on that, and share the update later on.

fujitatomoya, 2024-10-17:
https://github.com/ament/ament_lint/pull/505 worked like a silver bullet... 🦀

Crola1702, 2024-10-29:
Closing this one as the test is not failing anymore

InvincibleRMC, 2024-11-25:
This probably isn't related. But I noticed that the segfaults occur in `flake8-quotes` and that `flake-quotes` has [the](https://github.com/zheller/flake8-quotes/blob/3.3.1/requirements-dev.txt) restriction of less than v4.0 for `flake8` and that `flake8` is currently 4.0.1 in the CI.

Blast545, 2024-12-04:
:farmer: This started to happen reliably now, see: https://ci.ros2.org/job/nightly_linux-rhel_debug/2123/ and https://ci.ros2.org/job/nightly_linux-rhel_release/2098/. 

Re-opening. Probably https://github.com/ament/ament_lint/pull/505 can fix this.

PR #959 (2025-01-02): Rename the test_{daemon,direct}.py tests.
The reason for this is somewhat difficult to explain.

For many months now, we've had a regression in RHEL CI where Python would have a segmentation fault during the run of flake8 in the ros2cli package.  After a bunch of local debugging, I found that there is some crash in Fast-DDS teardown.  Specifically, sometimes it is attempting to join on a thread where the backing memory has already been erased somehow.  Despite a lot of debugging, and the use of asan and valgrind, I have not been able to find what is removing that memory before the join happens [... cut]
closed 2025-01-02 by clalancette