ros2/ros2cli #936: ros2 param dump does not work if a node has declared parameters without a value

reinzor, 2024-09-20:
## Bug report

**Required Info:**

- Operating System:

`Ubuntu 24.04`

- Installation type:

Debian package

- Version or commit hash:

```
0.32.1
```

- DDS implementation:

Cyclone

#### Steps to reproduce issue

```
python3 -c "import rclpy; rclpy.init(); node = rclpy.create_node('test'); node.declare_parameter('param', rclpy.Parameter.Type.INTEGER); rclpy.spin(node)"
```

```
ros2 param dump /test
```

#### Expected behavior

Dump node parameters

#### Actual behavior

UnboundLocalError: cannot access local variable 'e' where it is not associated with a value, should be fixed with https://github.com/ros2/ros2cli/pull/933

#### Additional information

Follow up of https://github.com/cra-ros-pkg/robot_localization/issues/903, related to https://github.com/ros2/ros2cli/issues/930

fujitatomoya, 2024-09-20:
@reinzor thanks for creating issue.

i just want to confirm that https://github.com/ros2/ros2cli/pull/933 can fix this issue as well, right?

reinzor, 2024-09-21:
No, it will not fix this behavior. It will only ensure that no UnboundLocalError is raised. Instead it will return an empty yaml. However, this is not correct.

fujitatomoya, 2024-09-23:
@reinzor 

so i tested with your example and https://github.com/ros2/ros2cli/pull/933 applied.

```bash
### your sample
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run prover_rclpy ros2cli_936 

### dumping parameters
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param dump /test
/test:
  ros__parameters: {}

```

it works as expected we have talked in https://github.com/ros2/ros2cli/pull/933#discussion_r1761685503?

reinzor, 2024-09-24:
Yes but this issue addresses another problem. 

```
python3 -c "import rclpy; rclpy.init(); node = rclpy.create_node('test'); node.declare_parameter('param', rclpy.Parameter.Type.INTEGER); rclpy.spin(node)"
```

Gives an empty output; but,

```
python3 -c "import rclpy; rclpy.init(); node = rclpy.create_node('test'); rclpy.spin(node)"
```

returns

```
/test:
  ros__parameters:
    start_type_description_service: true
    use_sim_time: false
```

So the problem is here that if we have one parameter without a value defined, the `ros2 param dump` command does not work. This is different from issue https://github.com/ros2/ros2cli/issues/930

fujitatomoya, 2024-09-24:
@reinzor 

Ah, yeah that is right. my expectation was wrong there...

So the thing is not related to `ros2cli` nor `ros2param` package, root cause why we end up having the empty parameter yaml file as output of `ros2 param dum` is https://github.com/ros2/rclcpp/issues/2512. that explains well, but in quick, the `GetParameter` service response will be empty if any parameter undeclared or no default values.

I think we can proceed this behavior change with https://github.com/ros2/rclcpp/issues/2512 (follow up needs to be done with `rclpy` for sure), so i will go ahead to close this issue. if you find specific issue on ros2cli, please feel free to reopen this.

closed 2024-09-24 by fujitatomoya

reinzor, 2024-09-25:
@fujitatomoya thanks for referring to that issue. Didn't know of its existence. Lets continue there!