ros2/ros2cli #937: [New Feature] Creating a flag for filtering package warnings using ros2 doctor

AngeLoGa, 2024-09-25:
----
## Feature request
Adding a new flag for clearing information of packages version using the tool `ros2 doctor`
#### Feature description
It would be interesting to filter the information related to packages displayed by the tool `ros2 doctor` using some kind of flag. If I want to know about my network state, I wouldn't like to see each time all the package checks and versions.

#### Implementation considerations
- Possible implementation: Some kind of RE to filter any warnings coming from the path `/opt/ros/{ROS_DISTRO}/lib/python3.12/site-packages/ros2doctor/api/package.py`
- PROs: Clarity and avoid undisered checks in the packages sourced.
- CONs: None (IMHO)

AngeLoGa, 2024-09-25:
Update after checking the code a bit: There are entry points for check and report, would it be feasible to have another one the same as check but without the `PackageCheck` entry?

Extract of setup.py (ros2doctor)
---
```
entry_points={
        'ros2cli.command': [
            'doctor = ros2doctor.command.doctor:DoctorCommand',
            'wtf = ros2doctor.command.doctor:WtfCommand',
        ],
        'ros2doctor.checks': [
            'PlatformCheck = ros2doctor.api.platform:PlatformCheck',
            'NetworkCheck = ros2doctor.api.network:NetworkCheck',
            'TopicCheck = ros2doctor.api.topic:TopicCheck',
            'QoSCompatibilityCheck = ros2doctor.api.qos_compatibility:QoSCompatibilityCheck',
            'PackageCheck = ros2doctor.api.package:PackageCheck',
        ],
        'ros2doctor.report': [
            'PlatformReport = ros2doctor.api.platform:PlatformReport',
            'RosdistroReport = ros2doctor.api.platform:RosdistroReport',
            'NetworkReport = ros2doctor.api.network:NetworkReport',
            'RMWReport = ros2doctor.api.rmw:RMWReport',
            'TopicReport = ros2doctor.api.topic:TopicReport',
            'QoSCompatibilityReport = ros2doctor.api.qos_compatibility:QoSCompatibilityReport',
            'PackageReport = ros2doctor.api.package:PackageReport',
        ],
```

fujitatomoya, 2024-09-25:
So what you are requesting here is to make the report without packages information? correct?

i think that is reasonable, but would like to discuss how we want to design this.

- `--exclude-packages, -ep` option: used with `--report` and `--report-failed` to exclude the package information collecting nor printing.
- `--system, -s` option: only print the information about the platform and ROS 2 system without package information.

any other thoughts?

AngeLoGa, 2024-09-25:
Any reason why the first option is conditioned by the `--report` flag? I haven't used the tool in real situations, but I imagine using `ros2 doctor` to check my network status rather than knowing if my packages are outdated. IMHO second option is interesting, but I think the key here is to know if we want to avoid showing unnecessary information, or to indicate implicitly the info you want to check. 

In fact, I first wanted to publish another issue due to a confussion I had when going through ros2 tutorials ([link here](https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Getting-Started-With-Ros2doctor.html#:~:text=Let%E2%80%99s%20examine%20your,n%3E%20checks%20passed)). From that fragment I was expecting to receive only the check status of the network, and not the information of packages. I thought it was a feature implemented in recent distributions but I read again the info and perhaps it was only focussing in the final part of the response of `ros2 doctor`.

To sum up, I think the second attempt is the goto since you normally want to know the least the better and just go for the thing you know you are having issues with, and if not, you can go with the `--report` flag and check everything (even to post in discourse such as Roboticks Stack Exchange).

fujitatomoya, 2024-09-25:
> Any reason why the first option is conditioned by the --report flag?

i am not original author this sub-command, but i think `ros2 doctor --report` is meant to be used to create the issue report on github, that means package version list are really important for reproducible procedure or developers.

AngeLoGa, 2024-09-25:
>i think ros2 doctor --report is meant to be used to create the issue report on github

Yes, I got that impression as well, but then why did you mention the use of the flag `--exclude-packages` with the report flag in your first comment?

>--exclude-packages, -ep option: used with --report and --report-failed

I would see totally fair to use `ros2 doctor --exclude-packages` as its own. Perhaps I misunderstood your first comment

fujitatomoya, 2024-09-25:
> I would see totally fair to use ros2 doctor --exclude-packages as its own. Perhaps I misunderstood your first comment?

yeah i was mistaken, sorry. this makes sense too.

fujitatomoya, 2024-10-14:
@AngeLoGa are you willing to consider the contribution for this? if not, i am going to work on this sometime later.

AngeLoGa, 2024-10-15:
I just rechecked the issue yesterday and saw the "help wanted" label. It is my first contribution so I would need some time to double check how to contribute, submit the request and merge. Please spare me a couple of days to work on it. Thanks in advance

fujitatomoya, 2024-10-15:
we really appreciate your support for the ROS community. if anything is up, please let us know.

AngeLoGa, 2024-10-15:
I have installed ROS 2 source in order to test this. Right now, I have included two flags, -fr or --full-report which works as --report does right now, and -pr or --partial-report to print everything except PackageVersion report. I would like to get some feedback from some users to see if this feature would be appreciated, or come up with another prompt for this purpose. I will link this issue to a Roboticks Stack Exchange post so people can give more feedback about this. At the moment, I just skip the format_print of PackageVersion if the flag partial_report is true. Tomorrow I would like to finish this changing the behaviour to not generate the report instead of not printing it.

fujitatomoya, 2024-10-15:
i was thinking more like `--exclude-packages` to exclude `PackageReport` and `PackageCheck` instances, that option can be applied to any command. we can keep this open to get more feedback!

@ros2/team

AngeLoGa, 2024-10-17:
Hey @fujitatomoya! Guess we are all alone on this hehe. So I have been able to follow your instructions, I created a new flag in the ``generate_reports``` function called exclude_packages to filter the PackageReport from the groups variable. The only issue is...the arg parser is defined as mutually exclusive, so we can't use two flags simultaneously using the command.

Double checking a couple of messages ago...
> I would see totally fair to use `ros2 doctor --exclude-packages` as its own. Perhaps I misunderstood your first comment?

I could consider that when you use this flag you are still requesting a report and do the same as the -r flag does, except that it changes the boolean to filter the package one. The issue with this is that the normal usage of ros2 doctor doesn't prompt a report, but just passes some checks. So I think that just using the flag --exclude-packages might not specify clearly that you are requesting a report? What are your thoughts? Also, are there any tests in the package in order to check that my changes do not affect the normal usage of the command (didn't see any)?

fujitatomoya, 2024-10-17:
@AngeLoGa i am thinking that `--exclude-packages` option can be applied to either `--report` or not. it just excludes the entrypoint to load `PackagesXXX` just not do any check or test for pacakges. does that make sense?

AngeLoGa, 2024-10-18:
Perhaps I didn't explain correctly in my last reply. In the ```doctor.py``` script:
```
class DoctorCommand(CommandExtension):
    """Check ROS setup and other potential issues."""

    def add_arguments(self, parser, cli_name):
        group = parser.add_mutually_exclusive_group(required=False)
```

Because of this, we can only use one flag at a time. For this reason, if we included ```--exclude-packages``` as a new flag, we wouldn't be able to choose to generate a report or a check, because ```ros2 doctor -r --export-excludes``` don't work because of the parser. The only possibility I came up with (without modifying the args exclusive term) is to assume that we want a report when we use ```--exclude-packages``` , since ```ros2 doctor``` does the package's checks, but it isn't a problem, since it doesn't print so many lines as ```ros2 doctor -r``` does.

fujitatomoya, 2024-10-18:
> Because of this, we can only use one flag at a time. For this reason, if we included --exclude-packages as a new flag, we wouldn't be able to choose to generate a report or a check, because ros2 doctor -r --export-excludes don't work because of the parser.

we do not need to add `--exclude-packages` option in that group arguments, so that we can use it. it is just like '--include-warnings' does. then we should have no problem to your concern?

AngeLoGa, 2024-10-20:
So, I have everything set for the pull request. I have made a fork and also pushed the changes into my local version ([link here](https://github.com/AngeLoGa/ros2cli_ros2doctor_push/tree/rolling)). Since this is my first time contributing and doing pr. Should I use the ```rolling``` or ``jazzy``` branch? At the same time, should I create a different branch from the originals of ros2cli so you can merge it as it should? I tried to seek some guidelines about this, but I wasn't succesful finding any. Also, I'm sorry for all the questions, but I want to do things as they should, and be able to share how it should be done for future cases.

fujitatomoya, 2024-10-21:
yes that is correct, you can refer to https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing.html#contributing-code.

and `rolling` branch is the development branch, please target the `rolling` branch, then if necessary we can do backport. (since this is new option, probably we do not backport but technically we can.)

AngeLoGa, 2024-10-29:
Succesful merge into the ``````jazzy`````` branch. Thanks @fujitatomoya  for all the feedback and help and glad to have worked for the ROS community!

closed 2024-10-29 by AngeLoGa

fujitatomoya, 2024-10-29:
@AngeLoGa happy to help and contributions are always welcome! appreciate your effort.