ros2/ros2cli #951: [Question] - does ros2 topic echo support multiple fields

cwrx777, 2024-12-04:
Hi,

is it possible to echo more than one fields, e.g.
```bash
  ros2 topic echo /topic --field field_1 --field field_2
```

clalancette, 2024-12-04:
In the current code, it does not support this; it can only print out either the whole message, or a single field.

That said, I think it would be an OK feature to be able to select more than one.  If you'd like to contribute a PR to implement it, we'd be happy to review it.

CWAbhi, 2024-12-28:
@clalancette can i  work in this issue?

fujitatomoya, 2024-12-28:
@CWAbhi contributions are always welcome, thanks 👍

CWAbhi, 2024-12-29:
okay I ll raise a PR to solve this

CWAbhi, 2024-12-29:
i have raised a PR with the changes done #958 please check and if i need to edit or do any changes do tell me.

PR #964 (2025-01-24): support multiple fields in ros2topic echo
Added support for multiple fields in `ros2 topic echo`. Fixes #951
closed 2025-01-24 by fujitatomoya