ros2/ros2cli #952: ros2doctor: print more information about (FastDDS) RMW

rursprung, 2024-12-07:
## Feature request

#### Feature description
`ros2 doctor` currently does not print a lot of information on the RMW (except which RMW is being used).
thus it's not helpful when debugging issues related to the middleware.
e.g. when using the discovery server it does not print any information about relevant `ROS_*` environment variables being set.
furthermore it could e.g. point out that a discovery server is set but `ROS_SUPER_CLIENT` isn't (which would explain why `ros2` commands don't see anything even if nodes are running).

#### Implementation considerations

fujitatomoya, 2024-12-07:
I think it would be nice to print more details of rmw implementation for practice, this would be really helpful to understand the environment for debugging instead of asking questions. for doing that, we probably would want to consider rmw interfaces to abstract to get the information from rmw implementation.

fujitatomoya, 2024-12-19:
@rursprung I had a chat about this with other developers.

what we could do is that develop ros2doctor plugin subcommand and get the user specific (or rmw specific) information instead of rmw interface. i think this is more reasonable.

IMO,

1. `ROS_xxx` / `RMW_xxx` environmental variables (ROS / RMW generic information) can be collected via ros2doctor. (we already do this with `RMW MIDDLEWARE` as report)
2. For rmw implementation specific parameters, user developed plugin can collect whatever those are. for example, `FASTDDS_BUILTIN_TRANSPORTS`, xml configuration data and so on.

maybe we can consider `ROS / RMW generic information` to expand the ros2doctor. but i would like to get more feedback on this.