ros2/ros2cli #954: Bug in ros2node for showing action servers

julianmueller, 2024-12-12:
Hi,
the following line causes an error for me:

https://github.com/ros2/ros2cli/blob/acefd9c0d773e7a067a6c458455eebaa2fbc6751/ros2node/ros2node/api/__init__.py#L128C28-L128C69

The called function `get_action_server_names_and_types_by_node()` does not actually exist here in `rclpy.node`, see: https://github.com/ros2/rclpy/blob/rolling/rclpy/rclpy/node.py

clalancette, 2024-12-12:
Can you please tell us what version of ROS 2 you are using, and the steps you took to get the error?  It looks to me like this exists on Rolling, but it is possible that we missed a backport or something.

julianmueller, 2024-12-12:
I'm using the current LTS, ROS2 Jazzy.  Where do you find the function in the rclpy.node module?

clalancette, 2024-12-12:
https://github.com/ros2/rclpy/blob/rolling/rclpy/rclpy/action/__init__.py#L18

Which is implemented in https://github.com/ros2/rclpy/blob/rolling/rclpy/rclpy/action/graph.py#L41 .
Which is actually actually implemented in https://github.com/ros2/rclpy/blob/rolling/rclpy/src/rclpy/node.cpp#L549 .

julianmueller, 2024-12-12:
Ah I see, thanks for the references!

But still, in https://github.com/ros2/ros2cli/blob/acefd9c0d773e7a067a6c458455eebaa2fbc6751/ros2node/ros2node/api/__init__.py#L128C28-L128C69 the function is called as a method of the `rclpy.node.Node` class, but it should rather be a function call from the `rclpy.action.graph` module

clalancette, 2024-12-13:
> But still, in https://github.com/ros2/ros2cli/blob/acefd9c0d773e7a067a6c458455eebaa2fbc6751/ros2node/ros2node/api/__init__.py#L128C28-L128C69 the function is called as a method of the `rclpy.node.Node` class, but it should rather be a function call from the `rclpy.action.graph` module

In theory, both should work, given that it is exported from the Node class.  I think at this point it would be helpful to know what commands you are running and what is not working for you.

julianmueller, 2024-12-15:
I called:
```python
get_action_client_info(node=ros2_node, remote_node_name=node_name, include_hidden=True)
```
with `ros2_node`as an instance of a ROS2 Node class, so a running node and get the python error:
```
AttributeError: 'Node' object has no attribute 'get_action_client_names_and_types_by_node'. Did you mean: 'get_client_names_and_types_by_node'?
```
This supports my previous claim, that the Node class does not have the method `get_action_server_names_and_types_by_node`.

My suggested fix would be:
```python
def get_action_server_info(*, node, remote_node_name, include_hidden=False):
    remote_node = parse_node_name(remote_node_name)
    names_and_types = get_action_server_names_and_types_by_node(
        node, remote_node.name, remote_node.namespace)
    return [
        TopicInfo(
            name=n,
            types=t)
        for n, t in names_and_types if include_hidden or not _is_hidden_name(n)]
```
and same for the `get_action_client_info()`

christophebedard, 2024-12-16:
How are you building or running ROS 2? Which distro are you using? If you're building ROS 2 from source, which branches are you using?

fujitatomoya, 2024-12-16:
@julianmueller 

> with ros2_nodeas an instance of a ROS2 Node class, so a running node and get the python error:

of course not, there is no such method in `Node` class.

> that the Node class does not have the method get_action_server_names_and_types_by_node.

No it does not. but `ros2cli` does not actually uses the `Node` class, it uses `NodeStrategy`. (that is why there is no problem in `ros2cli` commands)

but i understand your claim here, the following implementation generates the confusion...
and it should be implemented as `rclpy.graph` API to avoid these special cases.

https://github.com/ros2/ros2cli/blob/acefd9c0d773e7a067a6c458455eebaa2fbc6751/ros2cli/ros2cli/node/direct.py#L67-L79

it would be appreciated if you could consider the contribution to address above TODO.

julianmueller, 2024-12-23:
Thanks @fujitatomoya ! 
Maybe also type annotations and docstrings would be helpful, to know what type which argument should be and what it does. I'd be glad to contribute to that, if wanted.

clalancette, 2024-12-23:
> Maybe also type annotations and docstrings would be helpful, to know what type which argument should be and what it does. I'd be glad to contribute to that, if wanted.

Yep, that would be great.  We've been (slowly) adding type annotations to the entire ROS 2 codebase, but as there is a lot of code it has taken quite a while.  If you'd like to contribute to this effort, we'd be happy to review it.  I just ask that you do small PRs adding docs or annotations, rather than one giant PR.  Small PRs are easier to review and run CI for.

clalancette, 2026-11-14:
I believe that the original bug here was fixed by #1188 , so I'm closing this out.  Feel free to reopen or open a new issue if there is remaining work to do here.

closed 2026-11-14 by clalancette