ros2/ros2cli #960: bugs in `load_parameter_file`

huweiATgithub, 2025-01-12:
## Bug report

**Required Info:**

- Version or commit hash:
https://github.com/ros2/ros2cli/blob/28d73d257f232f81c8b2cd05c64cf29f10f272e8/ros2param/ros2param/api/__init__.py#L59-L65

#### Steps to reproduce issue
The current implementation has an extra `yaml.safe_load` for parameter value which has been loaded once already.
This will result in wrong results, e.g.,
```
key: ''
```
is once being loaded to `{"key": ""}`. Another load for the empty string `""` will convert it into `None`.

#### Implementation considerations
Consider removing the extra yaml load.

fujitatomoya, 2025-01-13:
@huweiATgithub can you provide the complete procedure (command line steps) that explain the problem here?

huweiATgithub, 2025-01-13:
> @huweiATgithub can you provide the complete procedure (command line steps) that explain the problem here?

1. having a node with a string type parameter
2. load parameter from a file using `ros2 param load` with file:
   ```yaml
   /**:
     ros__parameters:
       param_name: ''
   ```
3. Notice an exception.

fujitatomoya, 2025-01-13:
using https://github.com/fujitatomoya/ros2_test_prover/commit/1fcc859c964e3db3fd927385c47a81091c560717 and the following procedure that you provided above, it does not generate the exception.

```bash
### start the node with string parameter
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run prover_rclpy ros2cli_960 
[INFO] [1736797972.379024312] [minimal_param_node]: Got my_parameter: 
[INFO] [1736798063.849885193] [minimal_param_node]: Got my_parameter: 

### set the string parameter with ros2 param load
root@tomoyafujita:~/ros2_ws/colcon_ws# cat param_string_empty.yaml 
/**:
  ros__parameters:
    my_parameter: ''

root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param load minimal_param_node param_string_empty.yaml 
Set parameter my_parameter successful
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get /minimal_param_node my_parameter
String value is: 
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param describe /minimal_param_node my_parameter
Parameter name: my_parameter
  Type: string
  Constraints:
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param load minimal_param_node param_string_empty.yaml 
Set parameter my_parameter successful
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get /minimal_param_node my_parameter
String value is: 
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param describe /minimal_param_node my_parameter
Parameter name: my_parameter
  Type: string
  Constraints:
```

i would like to ask you again, could you please provide the [Short, Self Contained, Correct (Compilable), Example](https://sscce.org/)?

huweiATgithub, 2025-02-18:
> using [fujitatomoya/ros2_test_prover@1fcc859](https://github.com/fujitatomoya/ros2_test_prover/commit/1fcc859c964e3db3fd927385c47a81091c560717) and the following procedure that you provided above, it does not generate the exception.
> 
> ### start the node with string parameter
> root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run prover_rclpy ros2cli_960 
> [INFO] [1736797972.379024312] [minimal_param_node]: Got my_parameter: 
> [INFO] [1736798063.849885193] [minimal_param_node]: Got my_parameter: 
> 
> ### set the string parameter with ros2 param load
> root@tomoyafujita:~/ros2_ws/colcon_ws# cat param_string_empty.yaml 
> /**:
>   ros__parameters:
>     my_parameter: ''
> 
> root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param load minimal_param_node param_string_empty.yaml 
> Set parameter my_parameter successful
> root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get /minimal_param_node my_parameter
> String value is: 
> root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param describe /minimal_param_node my_parameter
> Parameter name: my_parameter
>   Type: string
>   Constraints:
> root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param load minimal_param_node param_string_empty.yaml 
> Set parameter my_parameter successful
> root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get /minimal_param_node my_parameter
> String value is: 
> root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param describe /minimal_param_node my_parameter
> Parameter name: my_parameter
>   Type: string
>   Constraints:
> i would like to ask you again, could you please provide the [Short, Self Contained, Correct (Compilable), Example](https://sscce.org/)?

Sorry for that I forget to mention the distro is `humble`.

The issue is in the function `get_parameter_value` which was moved to `rclpy` and got fixed in https://github.com/ros2/rclpy/commit/aeed14308a51b30eebf1fddd59a1d68389b38375.

The difference can also be viewed by comparing:
- https://github.com/ros2/ros2cli/blob/9e71d508418d2226618a76a0ece867137d3ff48a/ros2param/ros2param/api/__init__.py#L92-L94
- https://github.com/ros2/rclpy/blob/87fbec0d6bbfda968d14689c977e9a6bdaa48886/rclpy/rclpy/parameter.py#L271-L273

This can be verified as follows:

In humble with exception:
```
docker run --rm -ti ros:humble-ros-base bash
root@cf3e8cf151c9:/# python3
Python 3.10.12 (main, Jan 17 2025, 14:35:34) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from ros2param.api import get_parameter_value
>>> get_parameter_value("")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: get_parameter_value() takes 0 positional arguments but 1 was given
>>> get_parameter_value(string_value="")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/api/__init__.py", line 94, in get_parameter_value
    value.string_value = yaml_value
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rcl_interfaces/msg/_parameter_value.py", line 239, in string_value
    assert \
AssertionError: The 'string_value' field must be of type 'str'
```

In jazzy:
```
docker run --rm -ti ros:jazzy-ros-base-noble bash
root@748f8acac8f0:/# python3
Python 3.12.3 (main, Jan 17 2025, 18:03:48) [GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from rclpy.parameter import get_parameter_value
>>> get_parameter_value(string_value="")
rcl_interfaces.msg.ParameterValue(type=4, bool_value=False, integer_value=0, double_value=0.0, string_value='', byte_array_value=[], bool_array_value=[], integer_array_value=[], double_array_value=[], string_array_value=[])
```

fujitatomoya, 2025-02-26:
@huweiATgithub thank you for providing the information.

here is the fix https://github.com/ros2/ros2cli/pull/984, it would be appreciated if you can try to see if your problem can be fixed with it.

huweiATgithub, 2025-02-27:
@fujitatomoya Yes, it works!

closed 2025-02-27 by fujitatomoya