ros2/ros2cli #983: No need to deserialize the message via `hz` and `bw` unless necessary

fujitatomoya, 2025-02-24:
### Description

`hz` and `bw` verbs basically calculate the frequency and bandwidth for the messages on subscription.
basically it does not have to deserialize the message every time unless it needs to filter out the messages for the calculation.

for example, `ros2 topic hz --filter xxx`.

https://github.com/ros2/ros2cli/blob/f5e5a798d57c437410452f2b106f115d3a42c7a2/ros2topic/ros2topic/verb/hz.py#L70-L73

instead, it should receive the serialized messages and calculate the frequency and bandwidth for more performance especially large data message such as images.

### Implementation Consideration

For doing this, we need to develop `GenericSubscription` in `rclpy` to receive the serialized message in the user callback with `Node.create_generic_subscription` just like `rclcpp` does.

https://github.com/ros2/rclcpp/blob/rolling/rclcpp/include/rclcpp/node.hpp#L348-L374

and then call those APIs in `ros2topic` if it dose not need to receive the deserialized messages.

exMachina316, 2025-03-30:
I would like to work on this issue.

kscottz, 2025-04-01:
@exMachina316 I've assigned the issue to you. Please keep us to do date on your progress.

fujitatomoya, 2025-04-07:
@exMachina316 you are right, it does not required `GenericSubscription` to achieve this requirement. it would be nice to have but that is the another feature enhancement. i think we can enable `raw` when it can be, then keep it under `ROSTopicHz` and ignore the serialized message in the callback.

sorry for confusing you, i do not know why i came up with `GenericSubscription` idea before, maybe mixed up with other things 😓

fujitatomoya, 2025-04-07:
@exMachina316 thanks for pointing that out. i corrected suggested fix in the issue header.

PR #1005 (2025-04-09): Conditional deserialization of message for `ros2 topic hz`
Fixes #983 

1. Added a conditional statement to set the `raw` flag when creating subscriptions for topics based on the value of `filter_expr`. If the `--filter` flag is used, `filter_expr` is not None and deserializing subscriptions are created. Otherwise `filter_expr` is None and 'raw' subscriptions are created.

2. As for the `bw` verb, it already uses 'raw' subscriptions and there is no case where desrialization of topics is required. Hence no changes made there.

## Manual Testing
1. Run an example publisher:
```bash
$ ros2 run ex [... cut]
PR #1006 (2025-04-09): Conditional deserialization of message for `ros2 topic hz` (backport #1005)
Fixes #983 

1. Added a conditional statement to set the `raw` flag when creating subscriptions for topics based on the value of `filter_expr`. If the `--filter` flag is used, `filter_expr` is not None and deserializing subscriptions are created. Otherwise `filter_expr` is None and 'raw' subscriptions are created.

2. As for the `bw` verb, it already uses 'raw' subscriptions and there is no case where desrialization of topics is required. Hence no changes made there.

## Manual Testing
1. Run an example publisher:
```b [... cut]
closed 2025-04-09 by christophebedard