ros2/ros2cli #1002: `_rostopic_hz() got an unexpected keyword argument 'qos_args'`

mjforan, 2025-04-04:
### Operating System:

Linux 7789fa2838e0 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

### ROS version or commit hash:

jazzy

### RMW implementation (if applicable):

rmw_fastrtps_cpp

### 'ros2 doctor --report' output

```
   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<LOOPBACK,RUNNING,UP>
mtu          : 65536
inet         : 172.17.0.3
inet4        : ['172.17.0.3']
ether        : ea:a8:23:8a:0d:c2
netmask      : 255.255.0.0
device       : eth0
flags        : 4163<MULTICAST,RUNNING,UP,BROADCAST>
mtu          : 1500
broadcast    : 172.17.255.255

   PACKAGE VERSIONS
rqt_bag_plugins                           : latest=1.5.4, local=1.5.4
rosbag2                                   : latest=0.26.6, local=0.26.6
rosbag2_storage_default_plugins           : latest=0.26.6, local=0.26.6
rosbag2_storage_mcap                      : latest=0.26.6, local=0.26.6
rosbag2_compression_zstd                  : latest=0.26.6, local=0.26.6
mcap_vendor                               : latest=0.26.6, local=0.26.6
zstd_vendor                               : latest=0.26.6, local=0.26.6
vision_msgs_rviz_plugins                  : latest=4.1.1, local=4.1.1
rviz_visual_testing_framework             : latest=14.1.8, local=14.1.8
rviz2                                     : latest=14.1.8, local=14.1.8
rviz_default_plugins                      : latest=14.1.8, local=14.1.8
rviz_common                               : latest=14.1.8, local=14.1.8
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_examples_py                       : latest=0.26.6, local=0.26.6
launch_testing_examples                   : latest=0.19.5, local=0.19.5
ros2bag                                   : latest=0.26.6, local=0.26.6
rosbag2_py                                : latest=0.26.6, local=0.26.6
rosbag2_transport                         : latest=0.26.6, local=0.26.6
rosbag2_storage_sqlite3                   : latest=0.26.6, local=0.26.6
rosbag2_examples_cpp                      : latest=0.26.6, local=0.26.6
rosbag2_compression                       : latest=0.26.6, local=0.26.6
rosbag2_cpp                               : latest=0.26.6, local=0.26.6
rosbag2_storage                           : latest=0.26.6, local=0.26.6
interactive_markers                       : latest=2.5.4, local=2.5.4
common_interfaces                         : latest=5.3.6, local=5.3.6
visualization_msgs                        : latest=5.3.6, local=5.3.6
dummy_robot_bringup                       : latest=0.33.5, local=0.33.5
robot_state_publisher                     : latest=3.3.3, local=3.3.3
kdl_parser                                : latest=2.11.0, local=2.11.0
urdf                                      : latest=2.10.0, local=2.10.0
urdf_parser_plugin                        : latest=2.10.0, local=2.10.0
turtlesim                                 : latest=1.8.3, local=1.8.3
trajectory_msgs                           : latest=5.3.6, local=5.3.6
tracetools_test                           : latest=8.2.3, local=8.2.3
tracetools_launch                         : latest=8.2.3, local=8.2.3
topic_statistics_demo                     : latest=0.33.5, local=0.33.5
topic_monitor                             : latest=0.33.5, local=0.33.5
[... 348 lines cut ...]

```
Traceback (most recent call last):
  File "/ros2_ws/install/ros2cli/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli', 'console_scripts', 'ros2')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ros2_ws/build/ros2cli/ros2cli/cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ros2_ws/build/ros2topic/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/ros2_ws/build/ros2topic/ros2topic/verb/hz.py", line 87, in main
    return main(args)
           ^^^^^^^^^^
  File "/ros2_ws/build/ros2topic/ros2topic/verb/hz.py", line 152, in main
    _rostopic_hz(node.node, topics, qos_args=args, window_size=args.window_size,
TypeError: _rostopic_hz() got an unexpected keyword argument 'qos_args
```

### Additional information

Probably related to #935 

ROS built from source in a Docker container.

christophebedard, 2025-04-04:
That function in Jazzy doesn't have the `qos_args` parameter: https://github.com/ros2/ros2cli/blob/f037c1961f2b108be734d20c8f770c5e9d523856/ros2topic/ros2topic/verb/hz.py#L297. Make sure you're building the `jazzy` branch from source, and not the default (`rolling`) branch.

christophebedard, 2025-04-04:
Ah, wait, but it does get called here with `qos_args=args`: https://github.com/ros2/ros2cli/blob/f037c1961f2b108be734d20c8f770c5e9d523856/ros2topic/ros2topic/verb/hz.py#L152.

christophebedard, 2025-04-04:
That was mistakenly introduced by this PR: https://github.com/ros2/ros2cli/pull/998

christophebedard, 2025-04-04:
> That was mistakenly introduced by this PR: [#998](https://github.com/ros2/ros2cli/pull/998)

I've added 2 comments to #998 describing 2 things that need to be fixed: https://github.com/ros2/ros2cli/pull/998#pullrequestreview-2744027072. I'll open a PR.

Thanks for reporting!

christophebedard, 2025-04-04:
Alright, I ended up reverting the commit: #1004. The actual issue you were facing here should be resolved (otherwise let us know!), so I'll close this.

closed 2025-04-04 by christophebedard