ros2/ros2cli #1013: 🧑‍🌾 test verb list and dump connection refused

Crola1702, 2025-04-23:
### Operating System:

Ubuntu 24.04 and RHEL

### ROS version or commit hash:

rolling

### RMW implementation (if applicable):

All

### Steps to reproduce issue

1. Run a build in nightly_linux_release.
2. See test regressions https://ci.ros2.org/view/nightly/job/nightly_linux_release/3442/

### Expected behavior

Tests should not fail

### Actual behavior

Test are failing

### Additional information

Reference builds:
* https://ci.ros2.org/view/nightly/job/nightly_linux_release/3442/
* https://ci.ros2.org/job/nightly_linux_coverage/27505
* https://ci.ros2.org/job/nightly_linux-aarch64_release/3079/

This test has a flaky behavior:

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
nightly_linux_humble_coverage|2025-04-23|2025-04-17|13|7|53.85
nightly_linux_coverage|2025-04-23|2025-04-17|13|4|30.77
nightly_linux-rhel_release|2025-04-23|2025-04-08|14|9|64.29
nightly_linux-rhel_debug|2025-04-23|2025-04-09|15|11|73.33
nightly_linux-aarch64_repeated|2025-04-23|2025-04-15|16|7|43.75
nightly_linux-aarch64_release|2025-04-23|2025-04-16|12|6|50.0
nightly_linux_debug|2025-04-22|2025-04-15|14|6|42.86
nightly_linux-rhel_repeated|2025-04-22|2025-04-08|14|9|64.29
Rci__nightly-connext_ubuntu_noble_amd64|2025-04-22|2025-04-18|5|5|100.0
nightly_linux_release|2025-04-20|2025-04-15|12|6|50.0
nightly_linux-aarch64_debug|2025-04-20|2025-04-17|13|3|23.08
nightly_linux_repeated|2025-04-19|2025-04-18|12|2|16.67

fujitatomoya, 2025-04-24:
I cannot reproduce this issue with rolling source build / ubuntu noble. I do not get to the root cause of this issue, so the following is just a note.

https://ci.ros2.org/view/nightly/job/nightly_linux_release/3442/testReport/ros2param.ros2param.test/test_verb_list/test_verb_list/ (https://ci.ros2.org/view/nightly/job/nightly_linux_release/3442/testReport/junit/ros2param.ros2param.test/test_verb_dump/test_verb_dump/ looks almost same.)
- This test starts the parameter node test fixture and restart the `ros2daemon` before each test.
- This test uses `NodeStrategy`, it is likely that it uses `DaemonNode` since `ros2daemon` is restarted in the test. But if something happens to `ros2daemon`, it falls back to `DirectNode`. (`NodeStrategy` checks corresponding XMLRPC connection to the daemon every time by using `getattr`.)
- `ConnectionRefusedError: [Errno 111] Connection refused` exception indicates that something happened to the daemon when it is calling `get_service_names_and_types_by_node`, but test does not catch the exception. I think we can catch this `ConnectionRefusedError` in the `setUp`, that gives the retry to fall back to use `DirectNode` to keep the test running.
- timeout of `setUp` indicates that maybe the parameter node test fixture is not running anymore?

fujitatomoya, 2025-05-01:
> This test uses NodeStrategy, it is likely that it uses DaemonNode since ros2daemon is restarted in the test. But if something happens to ros2daemon, it falls back to DirectNode. (NodeStrategy checks corresponding XMLRPC connection to the daemon every time by using getattr.)

https://github.com/ros2/ros2cli/pull/1014 only mitigates the failure, but addresses the root cause of this issue, which is test fixture parameter node is not started as expected.

@Crola1702 can we see if the nightly builds becomes more stable for a bit?

Crola1702, 2025-06-10:
@cottsay do you think this can be solved by adding rmw_test_fixture?