ros2/ros2cli #1015: ros2 pkg create: Add --message and --service

peci1, 2025-04-25:
### Description

It would be nice if `ros2 pkg create` would also have `--message` and `--service` in addition to `--node-name` and `--library-name`.

### Motivation

It would be great if `pkg create` could generate a properly configured package with messages and services.

### Design / Implementation Considerations

When `--message` or `--service` is used, package type has to be `ament_cmake`.

mjcarroll, 2025-04-25:
> package type has to be ament_cmake

One of the roadmap items we are considering for L-turtle is rosidl enhancements to break the dependence on cmake for message generation.  Nothing committed yet, but FYI.

christophebedard, 2025-04-25:
> It would be nice if `ros2 pkg create` would also have `--message` and `--service` in addition to `--node-name` and `--library-name`.

This could look like: `ros2 pkg create ... --message MyMsg OtherMsg --service MySrv --action MyAction`

That would create:

* `msg/MyMsg.msg`
* `msg/OtherMsg.msg`
* `srv/MySrv.srv`
* `action/MyAction.action`

> It would be great if `pkg create` could generate a properly configured package with messages and services.

Then we could update these tutorials:

1. https://docs.ros.org/en/rolling/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html
2. https://docs.ros.org/en/rolling/Tutorials/Beginner-Client-Libraries/Single-Package-Define-And-Use-Interface.html
3. https://docs.ros.org/en/rolling/Tutorials/Intermediate/Creating-an-Action.html

Innomer, 2025-07-03:
> > package type has to be ament_cmake
> 
> One of the roadmap items we are considering for L-turtle is rosidl enhancements to break the dependence on cmake for message generation. Nothing committed yet, but FYI.

Just for clarification, do you mean to say something like a rosidl_cli or generator is being considered to break cmake dependencies?