ros2/ros2cli #1042: Incompatibility with EmPy 4.0

techtasie, 2025-06-03:
### Operating System:

```
<PUT OS INFO HERE>
```

### ROS version or commit hash:

jazzy

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

1. install empy 4.0
2. run ```ros2 pkg create test --build-type ament_cmake  --dependencies rclcpp std_msgs```
3. see: 
```
ros2 pkg create test \
  --build-type ament_cmake \
  --dependencies rclcpp std_msgs
going to create a new package
package name: test
destination directory: /home/timwendt/tmp
package format: 3
version: 0.0.0
description: TODO: Package description
maintainer: ['timwendt <techtasie@gmail.com>']
licenses: ['TODO: License declaration']
build type: ament_cmake
dependencies: ['rclcpp', 'std_msgs']
creating folder ./test
creating ./test/package.xml
Traceback (most recent call last):
  File "/usr/lib64/ros2-jazzy/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.32.4', 'console_scripts', 'ros2')())
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/lib64/ros2-jazzy/lib/python3.13/site-packages/ros2cli/cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
  File "/usr/lib64/ros2-jazzy/lib/python3.13/site-packages/ros2pkg/command/pkg.py", line 37, in main
    return extension.main(args=args)
           ~~~~~~~~~~~~~~^^^^^^^^^^^
  File "/usr/lib64/ros2-jazzy/lib/python3.13/site-packages/ros2pkg/verb/create.py", line 184, in main
    create_package_environment(package, args.destination_directory)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/ros2-jazzy/lib/python3.13/site-packages/ros2pkg/api/create.py", line 99, in create_package_environment
    _create_template_file(
    ~~~~~~~~~~~~~~~~~~~~~^
        'package_environment',
        ^^^^^^^^^^^^^^^^^^^^^^
    ...<2 lines>...
        'package.xml',
        ^^^^^^^^^^^^^^
        package_xml_config)
        ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/ros2-jazzy/lib/python3.13/site-packages/ros2pkg/api/create.py", line 81, in _create_template_file
    _expand_template(template_path, template_config, output_file_path)
    ^~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/ros2-jazzy/lib/python3.13/site-packages/ros2pkg/api/create.py", line 31, in _expand_template
    em.BUFFERED_OPT: True,
    ^^^^^^^^^^^^^^^
AttributeError: module 'em' has no attribute 'BUFFERED_OPT'
```

### Expected behavior

Create the files

### Actual behavior

Only creates the folder

### Additional information

I think the issue is similar to the one in [rosidl-#779](https://github.com/ros2/rosidl/issues/779).

I fixed it locally with this patch:
```diff
diff --git a/ros2pkg/ros2pkg/api/create.py b/ros2pkg/ros2pkg/api/create.py
index 98cc139..afdd1f4 100644
--- a/ros2pkg/ros2pkg/api/create.py
+++ b/ros2pkg/ros2pkg/api/create.py
@@ -42,12 +42,12 @@ def _expand_template(template_file, data, output_file):
             dispatcher=False,
             globals=data)
     else:
+        config = em.Configuration()
+        config.bufferedOutput = True
+        config.rawErrors = True
         interpreter = em.Interpreter(
             output=output,
-            options={
-                em.BUFFERED_OPT: True,
-                em.RAW_OPT: True,
-            },
+            options=config,
             globals=data)

     with open(template_file, 'r') as h:

```

fujitatomoya, 2025-06-03:
@techtasie this should be fixed by https://github.com/ros2/ros2cli/pull/921, can you confirm?

techtasie, 2025-06-06:
@fujitatomoya Yes it does. GitHub search didn't return this PullRequest for "empy". 

Thanks 👍

closed 2025-06-06 by techtasie