ros2/ros2cli #1061: 🧑‍🌾  ros2cli daemon get publishers and subscribers by topic test failing in connextdds

Crola1702, 2025-06-30:
### Operating System:

Ubuntu noble

### ROS version or commit hash:

Rolling and Kilted

### RMW implementation (if applicable):

rmw_connextdds

### Steps to reproduce issue

1. Run a build in Rolling or Kilted connextdds
2. See test regression fail

### Expected behavior

Tests should pass

### Actual behavior

Tests are failing

### Additional information

Reference builds:
- [Kci__nightly-connext_ubuntu_noble_amd64#17](http://build.ros2.org/job/Kci__nightly-connext_ubuntu_noble_amd64/17)
- [Rci__nightly-connext_ubuntu_noble_amd64#445](http://build.ros2.org/job/Rci__nightly-connext_ubuntu_noble_amd64/445)

Test regressions:
- [ros2cli.ros2cli.test.test_ros2cli_daemon.test_get_publishers_info_by_topic](https://build.ros2.org/job/Rci__nightly-connext_ubuntu_noble_amd64/445/testReport/junit/ros2cli.ros2cli.test/test_ros2cli_daemon/test_get_publishers_info_by_topic/)
- [ros2cli.ros2cli.test.test_ros2cli_daemon.test_get_subscriptions_info_by_topic](https://build.ros2.org/job/Rci__nightly-connext_ubuntu_noble_amd64/445/testReport/junit/ros2cli.ros2cli.test/test_ros2cli_daemon/test_get_subscriptions_info_by_topic/)

Comparing build repos diff points to #1040:

```diff
   ros2/ros2cli:
     type: git
     url: https://github.com/ros2/ros2cli.git
-    version: 9aec89715a62fca884f7e34d63456d2ee1b8a452
+    version: c0fabeaa7cdef097a92965104e3d809d9f6926f1
```

Log output:
```
test/test_ros2cli_daemon.py:232: in test_get_subscriptions_info_by_topic
    assert test_subscription_info.qos_profile.history == \
E   assert <HistoryPolicy.UNKNOWN: 3> == <HistoryPolicy.KEEP_LAST: 1>
E    +  where <HistoryPolicy.UNKNOWN: 3> = <rclpy.qos.QoSProfile object at 0x75cff096af10>.history
E    +    where <rclpy.qos.QoSProfile object at 0x75cff096af10> = <rclpy.topic_endpoint_info.TopicEndpointInfo object at 0x75cff0972500>.qos_profile
E    +  and   <HistoryPolicy.KEEP_LAST: 1> = <rclpy.qos.QoSProfile object at 0x75cff13a3370>.history
```

The reason it wasn't caught on CI in the PR is that CI runs with sequential executor, while nightlies run with parallel executor.

fujitatomoya, 2025-07-01:
@Crola1702 thanks for creating the issue for this problem.

i am not sure about the difference between sequential and parallel executor, but i can reproduce this issue with my local environment.

miguelgonrod, 2026-01-05:
Haven't happened again in a month, @Crola1702 can close the issue