ros2/ros2cli #1073: 🧑‍🌾 `ros2service.ros2service.test.test_cli.test_cli` failing consistently in Kilted

Crola1702, 2025-07-10:
### Generated by Generative AI

Part of it (using Claude Sonnet 4)

### Operating System:

Ubuntu 24.04

### ROS version or commit hash:

Kilted (but also flaky on Rolling)

### Steps to reproduce issue

1. Run a build in https://build.ros2.org/job/Kci__nightly-debug_ubuntu_noble_amd64/
2. See test regression fail

### Expected behavior

Test should not fail

### Actual behavior

Test is failing

### Additional information

Reference build: https://build.ros2.org/job/Kci__nightly-debug_ubuntu_noble_amd64/22/

Test regression: [ros2service.ros2service.test.test_cli.test_cli](https://build.ros2.org/job/Kci__nightly-debug_ubuntu_noble_amd64/22/testReport/junit/ros2service.ros2service.test/test_cli/test_cli/)

This is happening on cyclonedds, connextdds and fastrtps.

There are 8 expected services but 57 available services.

The tests expect only the intended test services and their associated parameter services, but the environment contains additional nodes.

This might be a problem of a contamination of the envrionment by problems with test isolation.

<details><summary>Test flakiness:</summary>

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
Jci__nightly-fastrtps-dynamic_ubuntu_noble_amd64|2025-07-09|2025-07-09|4|1|25.0
Rci__nightly-fastrtps_ubuntu_noble_amd64|2025-07-08|2025-07-08|12|1|8.33
Kci__nightly-release_ubuntu_noble_amd64|2025-07-08|2025-06-26|5|5|100.0
Kci__nightly-debug_ubuntu_noble_amd64|2025-07-07|2025-06-25|4|4|100.0
Kci__nightly-connext_ubuntu_noble_amd64|2025-07-07|2025-06-25|5|5|100.0
Rci__nightly-release_ubuntu_noble_amd64|2025-07-04|2025-06-29|12|3|25.0
nightly_linux_release|2025-06-25|2025-06-25|14|1|7.14
nightly_linux_debug|2025-06-25|2025-06-25|14|1|7.14
nightly_linux-aarch64_repeated|2025-06-25|2025-06-25|14|1|7.14
nightly_linux-aarch64_release|2025-06-25|2025-06-25|14|1|7.14
nightly_linux-aarch64_debug|2025-06-25|2025-06-25|13|1|7.69

</details>

<details><summary>Log output</summary>

```
=========================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================
FAIL: test_cli.TestROS2ServiceCLI.test_list_count[rmw_cyclonedds_cpp]
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/ws/install_isolated/launch_testing/lib/python3.12/site-packages/launch_testing/markers.py", line 61, in _wrapper
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/ws/src/ros2/ros2cli/ros2service/test/test_cli.py", line 210, in test_list_count
    assert int(output_lines[0]) == self.builtin_service_count + 1
AssertionError: assert 57 == (7 + 1)
[... 183 lines cut ...]
 +      and   ['/my_ns/echo_server/describe_parameters', '/my_ns/echo_server/get_parameter_types', '/my_ns/echo_server/get_parameter...y_ns/echo_server/get_type_description', '/my_ns/echo_server/list_parameters', '/my_ns/echo_server/set_parameters', ...] = <bound method TestROS2ServiceCLI.get_expected_builtin_services of <class 'test_cli.TestROS2ServiceCLI'>>('/my_ns/echo_server')
 +        where <bound method TestROS2ServiceCLI.get_expected_builtin_services of <class 'test_cli.TestROS2ServiceCLI'>> = <test_cli.TestROS2ServiceCLI testMethod=test_list_services[rmw_fastrtps_cpp]>.get_expected_builtin_services
 +    and   '/array_repeater/describe_parameters\n/array_repeater/get_parameter_types\n/array_repeater/get_parameters\n/array_repe...st_parameters\n/unbounded_sequences_repeater/set_parameters\n/unbounded_sequences_repeater/set_parameters_atomically\n' = <launch_testing.tools.process.ProcessProxy object at 0x7b81e5b9ca70>.output

=========================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================================
FAIL: test_cli.TestROS2ServiceCLI.test_list_with_types[rmw_fastrtps_cpp]
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/tmp/ws/install_isolated/launch_testing/lib/python3.12/site-packages/launch_testing/markers.py", line 61, in _wrapper
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/ws/src/ros2/ros2cli/ros2service/test/test_cli.py", line 194, in test_list_with_types
    assert launch_testing.tools.expect_output(
AssertionError: assert False
 +  where False = <function expect_output at 0x7b81efc5c220>(expected_lines=<itertools.chain object at 0x7b81e5bb2140>, text='/array_repeater/describe_parameters [rcl_interfaces/srv/DescribeParameters]\n/array_repeater/get_parameter_types [rcl...SetParameters]\n/unbounded_sequences_repeater/set_parameters_atomically [rcl_interfaces/srv/SetParametersAtomically]\n', strict=True)
 +    where <function expect_output at 0x7b81efc5c220> = <module 'launch_testing.tools' from '/tmp/ws/install_isolated/launch_testing/lib/python3.12/site-packages/launch_testing/tools/__init__.py'>.expect_output
 +      where <module 'launch_testing.tools' from '/tmp/ws/install_isolated/launch_testing/lib/python3.12/site-packages/launch_testing/tools/__init__.py'> = launch_testing.tools
 +    and   <itertools.chain object at 0x7b81e5bb2140> = <class 'itertools.chain'>(['/my_ns/echo [test_msgs/srv/BasicTypes]'], ['/my_ns/echo_server/describe_parameters [rcl_interfaces/srv/DescribeParameters]', '/my_ns/echo_server/get_parameter_t...ters [rcl_interfaces/srv/ListParameters]', '/my_ns/echo_server/set_parameters [rcl_interfaces/srv/SetParameters]', ...])
 +      where <class 'itertools.chain'> = itertools.chain
 +      and   ['/my_ns/echo_server/describe_parameters [rcl_interfaces/srv/DescribeParameters]', '/my_ns/echo_server/get_parameter_t...ters [rcl_interfaces/srv/ListParameters]', '/my_ns/echo_server/set_parameters [rcl_interfaces/srv/SetParameters]', ...] = <bound method TestROS2ServiceCLI.get_expected_builtin_services of <class 'test_cli.TestROS2ServiceCLI'>>('/my_ns/echo_server', with_types=True)
 +        where <bound method TestROS2ServiceCLI.get_expected_builtin_services of <class 'test_cli.TestROS2ServiceCLI'>> = <test_cli.TestROS2ServiceCLI testMethod=test_list_with_types[rmw_fastrtps_cpp]>.get_expected_builtin_services
 +    and   '/array_repeater/describe_parameters [rcl_interfaces/srv/DescribeParameters]\n/array_repeater/get_parameter_types [rcl...SetParameters]\n/unbounded_sequences_repeater/set_parameters_atomically [rcl_interfaces/srv/SetParametersAtomically]\n' = <launch_testing.tools.process.ProcessProxy object at 0x7b81e5bb2060>.output
```

</details>

fujitatomoya, 2025-09-25:
@Blast545 @Crola1702 for rolling, i think https://github.com/ros2/ros2cli/pull/1098 does the work.

fujitatomoya, 2025-09-25:
hmmm, i cannot reproduce this issue with kilted source build...

```console
root@tomoyafujita-B760M-Pro-RS-D4:~/ros2_ws/kilted_ws# for i in {1..10}; do
> colcon test --packages-select ros2service --pytest-args test/test_cli.py || break
> done
Starting >>> ros2service
[Processing: ros2service]
Finished <<< ros2service [32.8s]
...<snip>
```

as it reported, there seems to be some nodes running that are NOT supposed to be running during the test.

probably backporting rmw isolation features could resolve this flaky results?

Blast545, 2025-09-27:
🧑‍🌾 Those were flaky when there was Kilted was still rolling. I took a bit of time now to track if there was a clue as to when this stopped happening. I used the clue you said about this only happening when other node was still running and tried to check when the rolling jobs had a drop in the number of tests failing. 

It seems to me that backporting #1062 should do the trick. I will run a couple of tests for that.

Blast545, 2025-11-17:
This should be fixed after https://github.com/ros2/ros2cli/pull/1106 was merged, please track if those keep happening later this week @Crola1702 @miguelgonrod

miguelgonrod, 2025-12-04:
> This should be fixed after [#1106](https://github.com/ros2/ros2cli/pull/1106) was merged, please track if those keep happening later this week [@Crola1702](https://github.com/Crola1702) [@miguelgonrod](https://github.com/miguelgonrod)

That merge worked, this specific test isn't failing anymore in this builds. We are closing this issue

closed 2025-12-04 by Crola1702