ros2/ros2cli #1078: Deprecate `ros2 wtf`, use `ros2 doctor` only.

fujitatomoya, 2025-07-16:
### Description

Now `ros2 wtf` is alias of `ros2 doctor`, those are the same command line interface.
I believe this comes from ROS 1 history `roswtf`, but the functionalities between `roswtf` and `ros2 wtf` is different.
IMO, we can deprecate the `ros2 wtf` with deprecation period, and guide the user and developer to use official CLI `ros2 doctor` instead.
This is also good for maintenance.

### Motivation

See above.

### Design / Implementation Considerations

at least 1 distribution deprecation period should be given. that means, we can add the deprecation warning to `wtf` help message to suggest `ros2 doctor` instead. and we can remove that with next-Lyrical-Luth.

christophebedard, 2025-07-16:
I don't have a strong opinion, but `ros2 wtf` is kind of "fun" to have/keep.

asymingt, 2025-07-24:
We'll take this one to the PMC. Assigned myself to remember to add it to the agenda.

fujitatomoya, 2025-07-24:
@asymingt thanks for taking care of this!

InvincibleRMC, 2025-07-30:
The result of the discussion seemed to be to remove use of if from documentation but, otherwise keep it since the maintenance cost is extremely low since it is just an alias.

closed 2025-07-30 by fujitatomoya