ros2/ros2cli #1084: `ros2 --version` fails with "unrecognized arguments: --version" on Jazzy, Ubuntu 24.04

jamespyr, 2025-08-01:
### Generated by Generative AI

assist by Grok3

### Operating System:

Ubuntu 24.04 (Noble)

### ROS version or commit hash:

ROS 2 Version: Jazzy

### 'ros2 doctor --report' output

   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<RUNNING,LOOPBACK,UP>
mtu          : 65536
inet         : 192.168.160.84
inet4        : ['192.168.160.84']
ether        : ec:4c:8c:8e:f4:0e
inet6        : ['2402:7500:5ce:d785:d5ae:ced9:8089:4ee', '2402:7500:5ce:d785:5e49:7483:91ba:6734', 'fe80::f101:fc12:c270:f531%wlp0s20f3']
netmask      : 255.255.255.0
device       : wlp0s20f3
flags        : 4163<BROADCAST,RUNNING,MULTICAST,UP>
mtu          : 1500
broadcast    : 192.168.160.255
ether        : 38:a7:46:17:03:d2
device       : enp68s0
flags        : 4099<BROADCAST,MULTICAST,UP>
mtu          : 1500

   PACKAGE VERSIONS
ament_flake8                              : latest=0.17.2, local=0.17.2
rosidl_runtime_cpp                        : latest=4.6.5, local=4.6.5
ament_cmake_export_dependencies           : latest=2.5.4, local=2.5.4
rosidl_dynamic_typesupport                : latest=0.1.2, local=0.1.2
rviz_ogre_vendor                          : latest=14.1.13, local=14.1.12
rcpputils                                 : latest=2.11.2, local=2.11.2
rosgraph_msgs                             : latest=2.0.2, local=2.0.2
rosidl_dynamic_typesupport_fastrtps       : latest=0.1.0, local=0.1.0
std_msgs                                  : latest=5.3.6, local=5.3.6
ament_cppcheck                            : latest=0.17.2, local=0.17.2
ament_cmake_export_interfaces             : latest=2.5.4, local=2.5.4
unique_identifier_msgs                    : latest=2.5.0, local=2.5.0
rmw_implementation                        : latest=2.15.5, local=2.15.5
ament_uncrustify                          : latest=0.17.2, local=0.17.2
rosidl_typesupport_fastrtps_c             : latest=3.6.1, local=3.6.1
libstatistics_collector                   : latest=1.7.4, local=1.7.4
pcl_conversions                           : latest=2.6.2, local=2.6.2
ros2pkg                                   : latest=0.32.4, local=0.32.4
ament_cmake_cppcheck                      : latest=0.17.2, local=0.17.2
rosidl_runtime_py                         : latest=0.13.1, local=0.13.1
fastrtps_cmake_module                     : latest=3.6.1, local=3.6.1
rviz_rendering                            : latest=14.1.13, local=14.1.12
ament_cmake_core                          : latest=2.5.4, local=2.5.4
ament_cmake_version                       : latest=2.5.4, local=2.5.4
ament_lint_cmake                          : latest=0.17.2, local=0.17.2
pcl_msgs                                  : latest=1.0.0, local=1.0.0
ament_cmake_libraries                     : latest=2.5.4, local=2.5.4
ros2cli_common_extensions                 : latest=0.3.0, local=0.3.0
rosidl_parser                             : latest=4.6.5, local=4.6.5
ros2node                                  : latest=0.32.4, local=0.32.4
rcl_logging_spdlog                        : latest=3.1.1, local=3.1.1
ament_cmake_test                          : latest=2.5.4, local=2.5.4
rosidl_cli                                : latest=4.6.5, local=4.6.5
[... 167 lines cut ...]
- Working Commands: ros2 topic list, ros2 node list, ros2 service list, and example nodes (ros2 run demo_nodes_cpp talker and listener) function correctly.

- Python Module Checks:
1. python3 -c "import ros2cli; print(ros2cli.__file__)" outputs /opt/ros/jazzy/lib/python3.12/site-packages/ros2cli/__init__.py.
2. python3 -c "import ros2cli; print(ros2cli.__version__)" fails with AttributeError: module 'ros2cli' has no attribute '__version__'.
3. python3 -c "from ros2cli.cli import main; print(main.__version__)" fails with AttributeError: 'function' object has no attribute '__version__'.
4. python3 -c "from ros2cli.command import CommandExtension; print(CommandExtension)" outputs <class 'ros2cli.command.CommandExtension'>.

- Reinstallation Attempts:
1. Reinstalled ros-jazzy-ros2cli and ros-jazzy-ros2cli-common-extensions using:
         sudo apt-get remove ros-jazzy-ros2cli ros-jazzy-ros2cli-common-extensions
         sudo apt-get install ros-jazzy-ros2cli ros-jazzy-ros2cli-common-extensions
    The issue persists.
2. Reinstalled python3-pybind11 to address a metadata warning (WARNING: Skipping /usr/lib/python3.12/dist-packages/pybind11-2.11.1.dist-info due to invalid metadata entry 'name'), but the warning remains.

- Virtual Environment:
1. Created a Python virtual environment (python3 -m venv ~/ros2_venv) and installed argcomplete (3.6.2), pybind11 (3.0.0), and attempted to install ros2cli.
2. ros2cli used the system's /opt/ros/jazzy/lib/python3.12/site-packages/ros2cli due to source /opt/ros/jazzy/setup.bash, and ros2 --version still failed.

- Potential Cause:
1. The issue may be related to ros2cli’s command-line parsing (argparse) or compatibility with Python 3.12’s importlib.metadata (referenced in /opt/ros/jazzy/bin/ros2 script).
2. No directly related issues found in the ros2/ros2cli GitHub repository or ROS Discourse.

- Request
Could the maintainers confirm if this is a known issue with ros2cli 0.32.4 on Python 3.12? Are there recommended workarounds or patches for enabling ros2 --version functionality?

christophebedard, 2025-08-01:
`ros2 --version` doesn't exist.

There isn't really a "version" of ROS 2. If you're just trying to get the name of the current distro with a sourced workspace (e.g., `source /opt/ros/jazzy/setup.bash`), you can use the `ROS_DISTRO` environment variable:

```console
$ echo $ROS_DISTRO
jazzy
```

christophebedard, 2025-08-01:
Since this isn't supported and there's already a way to get the distro name, I'll close this. Feel free to keep discussing.